AIRLegend/aitrack #11: Receiving Application Error

dsaddons, 2020-08-10:
Followed the instructions and receiving this error
[image]

I am not seeing any video preview from my webcam within AITrack, the webcam works and is not being used by any other program.

Would appreciate some help, excited to try this out!

AIRLegend, 2020-08-10:
Hello and thanks for trying out AITrack!

This is a known bug related to some kind of problem with an internal library (still figuring out the root of the problem). 

As already reported by other users, try changing your USB camera to another port (USB 3.0, if possible). That seems to fix it.

dsaddons, 2020-08-10:
Hmm still receiving the same error.  It is a USB 3.0 webcam, have tried plugging it in to my USB 3.0 PCI card as well as the motherboard's USB 3.0 ports.  

I've deleted the program and redownloaded, AITrack still crashes but now no longer has an error that pops up.

AIRLegend, 2020-08-10:
Some questions, just to have more details:

- Have you downloaded the latest version? (v0.3)
- Have you tried connecting your webcam on different ports (3.0 and 2.0)?
- What camera do you have?
- What Windows version are you using?

dsaddons, 2020-08-10:
1. Yes, 0.3
2. I do not have any USB 2.0 ports unfortunately
3. Huddly GO
4. Windows 10 Pro
5. It crashes after clicking start tracking (which was the same way I received the original error)

AIRLegend, 2020-08-10:
Uhm, strange. everything seems okay... The only thing I can think of is  that your camera might have a compatibility issue during video capture. 

As I don't have that model to test, I would need to test that part. Could you please tell me if this works?:

1. Open Opentrack.
2. Select input "Aruco tracker".
3. Click track.
4. Check it displays video feed.

dsaddons, 2020-08-10:
Yep that did work

AIRLegend, 2020-08-10:
Okay, we can discard that then. Maybe executing the program as admin? 
Also make sure that all the original directories from the downloaded .zip are alongside AITrack.exe.
Maybe rebooting your machine could also fix the problem in case there is another thing interfering with the program.

dsaddons, 2020-08-10:
Have both restarted and ran AITrack as admin, no change.  I extracted AITrack into a folder and have left it unaltered.

I had a 2nd webcam until a few weeks ago when I gave it to a friend, so unfortunately no other webcam to test it out with.

AIRLegend, 2020-08-10:
That's very odd. Could you test it with another webcam? Maybe borrowing your friend's? Just to discard a problem with your machine.

kristankenney, 2020-08-10:
Just to add for troubleshooting, do you have any anti-malware/security software installed (Windows Defender or third party)?

dsaddons, 2020-08-10:
Unfortunately I do not have access to any other webcams due to COVID.  

My only AV is built in Windows.

AIRLegend, 2020-08-10:
I will review the code in charged of camera video handling. In the meantime, you could pass by the discord server, just to give you better support and notify you when the new version is released. Also, maybe another user could give you some hints... I had the first error you mentioned while building AITrack once, but changing my camera to another USB port fixed it...

biship, 2020-08-21:
Aitrack opens and then closes for me. No error.
I have to set Aruco in open track for it to display my Microsoft Q2F-00013 USB 2.0 LifeCam Webcam.

AIRLegend, 2020-08-21:
Make sure of the following things:

1. You don't have any other program using your camera when you launch AITrack (try not launching Opentrack with aruco tracker selected before AITrack).
2. AITrack.exe is located alongside it's needed files (.dll's, the models/ folder, etc).
3. You have read/write permissions on AITrack's folder. Try executing the program as admin.

If the above doesn't work, try changing your camera to another USB port to see if it still crashes.

AIRLegend, 2020-08-24:
New version should have improved this. Did it work for both of you?

dsaddons, 2020-08-25:
That fixed it for me!  Thank you for your help.  Excited to get a first
flight in.

AIRLegend, 2020-08-25:
This issue seems related to #20, which is already fixed. If the problem persists after the next update, please, report there.

Closing.

closed 2020-08-25 by AIRLegend