AIRLegend/aitrack #12: Unable to launch AITrack when Windows Defender ASR rule is enabled

kristankenney, 2020-08-12:
Attack Surface Reduction rule c1db55ab-c21a-4637-bb3f-a12568109d35 (Use advanced protection against ransomware) is blocking AITrack.exe from launching.

```
Microsoft Defender Exploit Guard has blocked an operation that is not allowed by your IT administrator.
 For more information please contact your IT administrator.
 	ID: C1DB55AB-C21A-4637-BB3F-A12568109D35
 	Detection time: 2020-08-12T16:49:31.248Z
 	User: ******\********
 	Path: C:\Users\*******\Downloads\aitracker-v0.3\AITrack.exe
 	Process Name: C:\Windows\explorer.exe
 	Security intelligence Version: 1.321.1239.0
 	Engine Version: 1.1.17300.4
 	Product Version: 4.18.2007.8
```

Windows 10 Pro for Workstations v2004, 19041.388.

https://docs.microsoft.com/en-us/windows/security/threat-protection/microsoft-defender-atp/attack-surface-reduction#use-advanced-protection-against-ransomware

AIRLegend, 2020-08-14:
Maybe executing  AITrack.exe as admin could fix the issue (?)

kristankenney, 2020-08-15:
Running as Administrator makes no difference in this case, and whitelisting AITrack.exe also has no effect. From what I can tell so far it looks like it's caused by one of the DLL's that the executable loads, will need to take a further look.

closed 2020-08-23 by AIRLegend

AIRLegend, 2020-08-24:
Could you confirm whether this is fixed with the new release (v0.5)?

PR #17 (2020-08-23): Bugfixes
- Changed IPResolver.cpp to use `127.0.0.1` as the fallback instead of the local IP address. This potentially resolves #12  because the program no longer makes TCP requests to external servers.
- Changed OCVCamera.cpp to resize each captured frame instead of forcing the camera to certain resolutions it could not handle (#16 ). (Mentioned by @rbroker and reported in #11).
- When user changed distance parameter in the GUI, PositionResolver.cpp was not updating this parameter right. This fixes the bug of the view jumping all over the place.

- The program now saves resolution (thanks @rb [... cut]