AIRLegend/aitrack #20: Crash on start up

knighty, 2020-08-24:
I've tried using both 0.4 and 0.5 and no matter what, every time I try and launch them the window (window only, no actual UI) shows up for a couple of seconds and crashes. I have a couple of error logs from Windows if that helps

```Faulting application name: AITrack.exe, version: 0.0.0.0, time stamp: 0x5f36c416
Faulting module name: ntdll.dll, version: 10.0.18362.815, time stamp: 0xb29ecf52
Exception code: 0xc0000005
Fault offset: 0x000000000001b38d
Faulting process ID: 0x1520
Faulting application start time: 0x01d67a39d4603851
Faulting application path: C:\Users\knigh\Downloads\aitracker-v0.4\aitracker-v0.4\AITrack.exe
Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report ID: 884315b6-c4cb-4221-a187-0c61f7e23e5f
Faulting package full name: 
Faulting package-relative application ID: 
```

```Fault bucket 1959026166248784696, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: AITrack.exe
P2: 0.0.0.0
P3: 5f36c416
P4: ntdll.dll
P5: 10.0.18362.815
P6: b29ecf52
P7: c0000005
P8: 000000000001b38d
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFF09.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFFB6.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFFD6.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFFF0.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER20.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_AITrack.exe_fee5552e8d284e2019772265c7d020aba0cc2fb_80129c8b_03dc7bda-dbe4-4cf1-b5bd-058f9f7ace09

Analysis symbol: 
Rechecking for solution: 0
Report Id: 884315b6-c4cb-4221-a187-0c61f7e23e5f
Report Status: 268435456
Hashed bucket: 048d30332d77101fab2fdbeb38f44338
Cab Guid: 0
```

AIRLegend, 2020-08-24:
Hmm... Looks like a memory error exeception. Just to get a little bit more information:

1. Have you installed  Visual C++ Redistributable x64 on your machine?
2. What camera do you have? Happens the same if you use Droid Cam with your mobile phone?
3. Are you launching AItrack.next to the other files included in the zip?
4. Check if Aitrack.exe has reading/write permissions (try executing as admin).

rafoloru, 2020-08-24:
I got the same problem here. 
- I installed C++ redis
- My Webcam is a Logitech HD Webcam C310
- I'm launching AItrack.exe from its original folder
- I tried to exacute as admin

Nothing of this steps works for me :(

knighty, 2020-08-24:
1. Yes I reinstalled with the linked one to check
2. Honestly I'm not sure what the webcam is, I've had it for many years. It's not the most ancient thing but it certainly isn't new. It's a logitech.
3. I extracted the folder and I'm launching the exe from that folder
4. Tried as an admin already, as well as with compatibility modes

AIRLegend, 2020-08-24:
Everything seems okay for you @rafoloru.

For you, @knighty, if the above doesn't work, please, test it with Droid Cam just to see if it still crashes and we can discard  a webcam problem.

As reported previously by other users when program crashed, changing the USB camera to another port (to a 3.0 one, if possible) seemed to work. Try that before Droid Cam.

knighty, 2020-08-24:
I tried putting in a usb 3.0 port and that made no difference. Trying to get DroidCam working at the moment with little luck but I'll let you know if/when I get this working

knighty, 2020-08-24:
I poked through searches for the signature a bit and when looking up `b29ecf52` I got this page. Could be something that points us somewhere?

https://answers.microsoft.com/en-us/windows/forum/all/integrated-web-cam-issue/03db172e-8a18-4cd6-9cb1-152fb3779011

AIRLegend, 2020-08-24:
Droid Cam is reported to work without issues. So I doubt the problem is camera related... However, just to check something more: Does your camera work using Opentrack with input = aruco tracker? (check you have video input in Opentrack).

NTTDLL is Windows related and, as using Droid Cam also makes the program crash, I'm suspecting it could be related to Windows itself.

Is it also possible that you have any program running already that is using the camera? Do you have any VR headset connected? (that could also confuse AITrack).

Arkhonist, 2020-08-24:
I have the same issue, I'm using Droid Cam and it does work with aruco tracker on opentrack

knighty, 2020-08-24:
Just checked and my camera is also a Logitech HD Webcam C310. It works in OBS. aruco tracker doesn't seem to show anything for me, not sure where I'd see video there.

AIRLegend, 2020-08-24:
I think there are already people using this particular webcam. I'm suspecting it's an issue related to Windows itself (but not sure).

I just told @Arkhonist  on Discord to update Windows to see if that solves the problem. What version are you using? Some users reported that updating solved the crashing problems.

knighty, 2020-08-24:
I actually just spent the last 2 hours doing updates because I had the same thought and I was 2 months behind because apparently my windows update broke, which led me through a whole process of resetting windows update to get it to run again. I've done all those updates now and still no luck unfortunately. The fact that I have no video input on opentrack with aruco seems suspicious to me. In case you were wondering, my privacy settings in windows are setup to allow anything to use the camera and verifying with OBS seems that it is the case as it works there

AIRLegend, 2020-08-24:
Good, a thing less to worry about.

Opentrack  doesn't receive video feed probably because it uses an almost deprecated (in windows) OpenCV video capture backend. AITrack tries to use that as a default but, in case of any error falls back to the most modern one (https://github.com/AIRLegend/aitrack/blob/684001779fd15d0a4d96ce8f7b35dd0bca8b28e7/Client/src/camera/OCVCamera.cpp#L9-L15), which should be working okay.

I'm quite sure there is any kind of conflict on a previous part of the program (before the camera initialization). 

Could you please come by to the discord server to deeply investigate that?

rbroker, 2020-08-24:
Just to link this up, the original issue seemed to be coming from libusb initializing the PS3 camera library:

```
     ntdll.dll!00007ffb0dd313ad()    Unknown
     ucrtbase.dll!00007ffb0b4d9ffe()    Unknown
     ucrtbase.dll!00007ffb0b4da11e()    Unknown
     AITrack.exe!usbi_log_v(libusb_context * ctx, libusb_log_level level, const char * function, const char * format, char * args) Line 2655    C
     AITrack.exe!usbi_log(libusb_context * ctx, libusb_log_level level, const char * function, const char * format, ...) Line 2668    C
     AITrack.exe!init_device(libusb_device * dev, libusb_device * parent_dev, unsigned char port_number, unsigned long devinst) Line 871    C
     AITrack.exe!winusb_get_device_list(libusb_context * ctx, discovered_devs * * _discdevs) Line 1486    C
     AITrack.exe!libusb_get_device_list(libusb_context * ctx, libusb_device * * * list) Line 839    C
>    AITrack.exe!ps3eye::USBMgr::listDevices(std::vector<std::shared_ptr<ps3eye::PS3EYECam>,std::allocator<std::shared_ptr<ps3eye::PS3EYECam>>> &) Line 404    C++
     AITrack.exe!ps3eye::PS3EYECam::getDevices(bool forceRefresh) Line 996    C++
     [External Code]  
```

We then ran into what seemed like a secondary issue, but I'm not sure if it's related to the original problem or not :(

rafoloru, 2020-08-25:
On my side,

- My Windows is up to date
- I tried with an other more recent webcam (on both USB 2 and USB 3)

And nothing working, the app still crash on start up

AIRLegend, 2020-08-25:
Confirming it's a problem related with the PS3 Camera library, as @rbroker suggests. I'm working on it.

rafoloru, 2020-08-25:
And BTW... Thanks for your work! :)

AIRLegend, 2020-08-25:
Problem fixed in https://github.com/AIRLegend/aitrack/commit/f177ca69e38637e5e3fd298f6f180e9decfbe6b9 and confirmed by @Arkhonist and @knighty. It will be out with the next release :) 
Closing.

closed 2020-08-25 by AIRLegend