AIRLegend/aitrack #21: PS3 Eye Driver

brian4gh, 2020-08-24:
Sorry for question but I have limited knowledge.  My camera contrast is very bright and I'm thinking I may need the PS3 experimental driver mentioned.  How do I get that and have it installed?  Right now I'm using CL-Eye Test driver.

AIRLegend, 2020-08-24:
Hello!
AITrack uses a built in PS3 driver, so the CL-Eye Test should not affect (I think). I personally use a PS3 camera and have no exposure problems (it uses the default camera settings). Have you tried changing the camera parameters inside the CL-Eye configuration utility? Maybe I'm wrong and that affects.

brian4gh, 2020-08-24:
I tried changing the CL-Eye Test exposure and it did not effect the video from Aitrack.  The video is very light contract (it may be designed that way) but when I move my head to the left only it loses track of my head.  If I close the blinds and make the room darker it works fine.

Best,

Brian

AIRLegend, 2020-08-24:
I will try to replicate that tomorrow. Will reach back to you!

brian4gh, 2020-08-24:
May have narrowed it down.  I uninstalled the CL_Eye test driver now I’m getting no tracking at all from AITrack.  Is there a new or different driver I need to installed for the PS3 Cam?

Brian

AIRLegend, 2020-08-24:
As far as I know, the CL-Eye is the only one which is free. That's the one I use, without any problems. Does it happen the same if you reinstall it?

brian4gh, 2020-08-24:
You just nailed it.  I literally just reinstalled the driver after manually cleaning it off my system and now it works great!  Thanks so much, I was about to buy TrackIR.  Glad I didn’t!  Works great with MFS 2020!

Best,

Brian

AIRLegend, 2020-08-24:
Great to hear that! That certainly was a very strange bug. Hope you like it!

(Closing issue).

closed 2020-08-24 by AIRLegend