AIRLegend/aitrack #22: Select camera?

Raging-Beard, 2020-08-24:
It's picking up the cam on my Vive instead of webcam. Any to change this please without unplugging the Vive? Thanks

AIRLegend, 2020-08-24:
Not at the moment, sorry. AITrack assumes as valid the first camera it finds. Maybe swapping the USB ports makes the trick, or, if not, disabling the Vive via Windows Device Manager would be a temporary solution.

Raging-Beard, 2020-08-24:
Ok thanks, yeah I just realised I can disable Vive camera in device manager which as done the job. Hopefully it shouldn't interfere with the Vive usage as I don't have the camera turned on in SteamVR anyway. Cheers.

skalteis, 2020-08-24:
Could we leave this feature request open, please?

I definitely would have use for a way to select the used camera since I am using two cameras (one for a video feed, second one for tracking). Even just specifying the DirectShow-ID or something similar via the configuration file would be perfectly fine for me, I would not even need the GUI. :-)

Thank you for this otherwise superb headtracking solution, I really appreciate the work. 👍

AIRLegend, 2020-08-24:
I will start working on it for the next update! :)

AIRLegend, 2020-08-29:
Already added in #34. Will be out with the next version!

closed 2020-08-29 by AIRLegend