AIRLegend/aitrack #24: Ps3 Eye Camera exposure

xsx, 2020-08-25:
Hello, first of all awesome program. It works flawlessly considering the limitations implied by the cameras used.

I have an issue with the integrated Ps3 driver your Aitrack is using, i think: I see that the exposure and gain seems fixed and not configurable, this causes the camera to be inoperable in dark environments.
Comparing the driver with the CL-Eye camera driver I see that in the CL-Eye Test app the camera is much more sensitive to light, allowing it to see my face in much darker environments, despite of obviously increased noise value, but this should not pose a problem to your tech. It is possible to bypass the integrated driver, or make the exposure and gain configurable by ini file in any way?

You should really open a donation something because you really deserve it.

Thanks.
Best regards
Silvio

AIRLegend, 2020-08-25:
Yes, they are fixed right now as I hadn't any problems in low light environments. Will add that option for the next version though!

AIRLegend, 2020-08-25:
Already started working on that and is available it in version 0.5.1. Until the GUI menu is added, you can edit the `prefs.ini` file to configure exposure/gain. Hope it improves your experience!

AIRLegend, 2020-08-25:
Marking as **enhancement** as it's still necessary to build the configuration menu.

AIRLegend, 2020-08-29:
Added this in #34 (also supporting generic cameras too as long as the camera itself can handle it). Will be out with the next version!

closed 2020-08-29 by AIRLegend