AIRLegend/aitrack #26: Black/Empty feed.

biship, 2020-08-25:
Latest Opentrack is running using Aruco and my camera works (and I see myself).
I start AITrack and it says No Video Input.
I click Start Tracking and get a black screen.
Using 0.5.1 (You really should put the ver number in the window title).

AIRLegend, 2020-08-26:
- What camera are you using?
- Did you have video feed in previous versions?

biship, 2020-08-26:
Microsoft Q2F-00013 USB 2.0 LifeCam Webcam (1080p)
No, all previous version crashed on startup.
0.5.1 is the only one to launch.

AIRLegend, 2020-08-26:
Is possible you have some program affecting AITrack? i.e. the antivirus blocking it's access to the camera. 
Also check, in Windows settings, that AITrack has camera permissions.

If nothing of the above works, please, come by the [discord server](https://discord.gg/HPZMdcx) and reach me so we can further investigate this problem.

biship, 2020-08-26:
I am only running Windows Defender, which I disabled for this test.
AITrack does not show in Windows Camera Settings.
Opentrack does show (and says Currently in use when showing the feed in the app).

AIRLegend, 2020-08-27:
I will take a look at it and report when I find the error.

isaacrlevin, 2020-08-27:
Is it possible you have 2 webcams? I have 3 (I know) and looking at ProcessMonitor it is stuck on one of them, even though it is unplugged. I imagine the way you are getting the camera (via the registry?) could be optimized

AIRLegend, 2020-08-27:
At the moment it only gets the first camera it sees 
https://github.com/AIRLegend/aitrack/blob/0078f28de3fc9131aae5742a019c83a168911d90/Client/src/camera/OCVCamera.cpp#L43

I'm going to add a camera selection feature though.

biship, 2020-08-27:
I only have one camera.
It shows up under Sound, Video and game controllers in the Device manager (next to the NVIDA & RealTek High Definition Audio devices).

AIRLegend, 2020-08-29:
After investigated the bug (thanks Wally), it has been confirmed that it's not produced by AITrack's codebase. 

Specifically, the error is produced when grabbing a frame in OpenCV 

https://github.com/opencv/opencv/blob/2e6e9c14b5bc689f9444836bc245956d690ea42b/modules/videoio/src/cap_dshow.cpp#L1553

Specifically, the error is  because other part of the code is not called by an external thread

https://github.com/opencv/opencv/blob/2e6e9c14b5bc689f9444836bc245956d690ea42b/modules/videoio/src/cap_dshow.cpp#L710
 
This is probably due to [some application (antivirus or windows rule) blocking the thread](https://stackoverflow.com/a/58630682/8230341) incharged of running the former code, so the returned frame is always empty.

Please check you don't have anything blacklisting AITrack.

closed 2020-08-29 by AIRLegend

biship, 2020-08-29:
lol. All other apps can access the camera, there isn't a AV running, nor is there any FW active.
So yeah, can't be AITrack.

AIRLegend, 2020-08-29:
@biship  Could you pass by the discord server so I can further review your case? It doesn't seem to be caused by that but it could be something similar. As I can't replicate the error it's difficult to debug it.

stottsc, 2020-08-31:
I think [this issue from Opentrack ](https://github.com/opentrack/opentrack/issues/611) may be related.

I have the same issue with black feed with a Facetime HD camera (Windows 10 1909, Boot Camp on iMac Pro). Camera app built into Windows works fine. Disabling AV (Windows Defender) made no change. If I use aruco or PointTracker in Opentrack, the same black feed issue occurs.

[This solution](https://github.com/opentrack/opentrack/issues/611#issuecomment-299715371) solved the problem for me, both for AITrack, and aruco and PointTracker in Opentrack. Perhaps this is a codec setting issue?

AIRLegend, 2020-08-31:
This could be related with this issue: https://github.com/AIRLegend/aitrack/issues/27#issuecomment-683853109

stottsc, 2020-08-31:
I had checked the Camera privacy settings as the first step in diagnosing this issue. The setting to allow desktop applications to access the camera was, and still is, enabled. If I disable the setting in Camera privacy, then AITrack quits on launch, and Opentrack doesn't show ANY feed from the camera, instead of showing a black feed.

Ousadanken, 2025-07-28:
5 years later and i have NO FUCKING CLUE WHY ITS BLACK STILL