AIRLegend/aitrack #27: No cameras detected

airick17, 2020-08-26:
When I open AiTrack it gives me a message saying no cameras detected. 
I have verified my webcam is working with both CL Eye Test and Open Track point tracker.

After that message it also displays a message saying there was a problem initializing the tracker. Check the model folders and restart the program.

Any ideas what I might be doing wrong? I have opentrack configured based on the examples provided.

AIRLegend, 2020-08-26:
For the camera, I suppose you're using a PS3 Eye camera. It should work out of the box. However, installing WinUSB drivers might do the trick (https://zadig.akeo.ie/ ).

The model error is thrown when AITrack can't reach the files inside `models/` folder. It should be next to your `AITrack.exe`. Also check that it has reading/write permissions (try adjusting them or executing as admin).

indiandennis, 2020-08-31:
I had the same errors with DroidCam, if the camera works when you test it but not with AITrack, it is likely due to a Windows privacy setting problem. I think Windows update to v2004 resets some settings because I'm pretty sure this wasn't a problem before. Go to your camera privacy settings and make sure you allow desktop apps to access your camera.

[image]
[image]

MisterFabulous, 2020-09-06:
Hi I have the same issue. I have a logitech streamcam which appears in the CAM app on Windows 10. When I open aitrack it says no camera detected. I have tried in configuration to set it to camera 0,1,2,3 but to no avail. I hope you can help as I am a novice at this. I have checked the settings and it is already allow desktop apps to access your camera.

AIRLegend, 2020-09-06:
What does `log.txt` contain?

heimann, 2020-09-06:
I have the exact same problem with a Logitech camera. 

The camera light flashes on for a second when I open the app then the light goes off. Preview never shows a feed. Privacy feed says aitrack.exe is using the camera.

Log output:
```
`[2020-09-06 12:50:48.616] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-06 12:50:48.898] [aitrack] [info] Created/Found prefs.ini
[2020-09-06 12:50:48.898] [aitrack] [info] Searching for cameras...
[2020-09-06 12:50:50.585] [aitrack] [info] Number of recognized cameras: 2
[2020-09-06 12:50:50.586] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-09-06 12:50:50.586] [aitrack] [info] Building Tracker with selected camera: 1
[2020-09-06 12:50:50.630] [aitrack] [info] Tracker initialized.
[2020-09-06 12:50:50.630] [aitrack] [info] Updated stabilizer.
[2020-09-06 12:50:50.630] [aitrack] [info] Updating camera parameters...
[2020-09-06 12:50:50.630] [aitrack] [info] Updated camera parameters. 640x480@0
[2020-09-06 12:50:50.630] [aitrack] [info] App initialized
[2020-09-06 12:50:52.780] [aitrack] [info] Starting camera 1 capture
[2020-09-06 12:50:52.888] [aitrack] [info] Camera 1 started capturing
```

AIRLegend, 2020-09-06:
@heimann  Check you don't have FPS=0 in the configuration window. If so, set it to 30.

heimann, 2020-09-06:
> 
> 
> @heimann Check you don't have FPS=0 in the configuration window. If so, set it to 30.

I did have it set to 0, but setting it to 30 doesn't change anything unfortunately. Still seeing "No active video feed".

AIRLegend, 2020-09-06:
@heimann So, the program stays the same (no crashing) and it still shows "No active video feed" after clicking start? What camera do you have? Can you please check if the error still exists when using Droid Cam?

heimann, 2020-09-06:
> 
> 
> @heimann So, the program stays the same (no crashing) and it still shows "No active video feed" after clicking start? What camera do you have? Can you please check if the error still exists when using Droid Cam?

Yeah it doesn't crash, it just shows a Camera with a slash through it. 

Same thing happens with DroidCam (over iOS). The webcam I'm using is a Logitech C920

MisterFabulous, 2020-09-06:
I get the same on my screen with 20fps and logitech streamcam
[image]

xEtarusx, 2020-09-09:
Same here.
Looks like my problem is coming from windows 10 privacy settings somehow. When I look into the camera access list as @indiandennis described there is the app AITrack.exe listed, but it says "Access denied" in red under it.
And I don't see any way to change this behavior. I've already allowed the camera access for apps in windows but it seems like windows doesn't care for this program.

AIRLegend, 2020-09-09:
Logitech cameras seem to be tricky to manage, however, if Droid Cam also doesn't work I think the problem can be due to some windows privacy rule or antivirus blocking the camera access. There was a case on prior versions of an user whose AITrack was accessing the camera right, but somehow, it's antivirus was  making it to read an empty video feed and that crashed the program. This sound similar to me. Could it be the case?

@xEtarusx  If you can't find a way to unblock it, I think renaming the .exe could do the trick.

Hambone70, 2020-09-09:
Logitech issue here, too. All privacy settings and antivirus are configured to allow AITrack access to the camera, but I also get "No active video feed". Norton acknowledges with "Allowed webcam access to AITrack.exe", but still no video feed.

Hambone70, 2020-09-10:
Got it working. Not too sure how, though...

AIRLegend, 2020-09-10:
@Hambone70 If you figure it out it, I'd appreciate you sharing what fixed it for other people with this problem. Was the AV?

xEtarusx, 2020-09-10:
@AIRLegend Renaming the .exe did not work for me. Any other Ideas? Got an old Logitech HD 720p camera and no logitech software installed for it.

AxGr70, 2020-09-11:
@AIRLegend Having the same issues with "No active video feed" and my Logitech C920 webcam as the other ones here. Tried DroidCam and Iriun Webcam with my phone also without any video showing up.

Running Windows 10 2004 with September Update, windows privacy settings give access to webcam, antivirus isn't blocking any webcam access either...

kognise, 2020-09-11:
I have a C920 as well, same issue - flashes blue when AITrack opens but then the light goes off and the app says "No active video feed"

WF000, 2020-09-23:
.. just FYI , I've had luck working after replugging PS3eye to an USB2 (intel) port - so if you still have one of those .. lol
 in USB3 it didn't work , maybe even because USB3 is on usb hub... 

BUT!, . OpenTrack is working over usb3 with no problem.. so I guess it's AiTrack's  problem accessing device via usb3 ports.. even with windows camera support disabled

I'm using "original" CL-Eye-Driver-5.3.0.0341-Emuline.exe  driver (the paid one)

unfortunately , experience crashing after cca- 5-10min .. so, not stable enough

[image]

t-bo, 2021-03-28:
Hi,
I confirm USB3 port was not working but USB2 is fine. Thank you

sb0, 2021-09-18:
My computer has no USB2 ports.  Is there another solution for this issue?
I have AITrack allowed in Windows privacy settings and OpenTrack is able to detect and see video feed.

searching46dof, 2022-06-30:
usb 3.0 ports provide more power than usb 2.0 ports and the camera power requirements may exceed a usb 2.0 port. I could hear windows installing a driver but a camera never shows up in device manager. you can also connect it indirectly to the computer via a powered usb hub.

if there is a problem where AITrack can't access files or folders which exist, then try setting an exclusion in the antivirus/antimalware product you are using.

mvrk69, 2023-09-24:
I have same problem, aitrack doesn't detect my camera (https://www.amazon.com/Nulaxy-C900-Microphone-Streaming-Conferencing/dp/B08D39BC1T), also plugged on USB3 ports, my PC doesn't have USB2.

opentrack and other apps detect the camera without problem.

vissermake, 2023-10-08:
> I have same problem, aitrack doesn't detect my camera (https://www.amazon.com/Nulaxy-C900-Microphone-Streaming-Conferencing/dp/B08D39BC1T), also plugged on USB3 ports, my PC doesn't have USB2.
> 
> opentrack and other apps detect the camera without problem.
> 
> Meanwhile i went back to aitrack 0.6.5, and it detects the camera, takes a lot of time to load and start tracking but it works, but with all versions after 0.6.5 it won't detect my camera.

Same Here change to  AItrack 0.6.5 worked for me but AItrack 0.6.6 and AItrack 0.7.0 don't work.

Ousadanken, 2024-04-22:
so what is the intial fix for this? i tried previous versions. not working am i doing somehtign wronggg???

Drago7092, 2025-01-05:
> After spending numerous hours finding a problem, why my android cam is not detected, i simply changed from droidcam to Iriun web cam app and it worked. Maybe this helps someone...

Same for me

nickarmoore, 2025-01-09:
i have no camera detected i allowed cameras in windows settings its working on droid app and obs camera is synced and udp over network is on what am I missing?
[image]
[image]

Ousadanken, 2025-07-28:
im gonna kms from this, even a year later and a different pc, still not working