AIRLegend/aitrack #28: Improve exception safety with smart pointers

elfring, 2020-08-26:
Would you like to [wrap any pointer data members](https://github.com/AIRLegend/aitrack/blob/9e455e10990969a157cbd1c066f90eede60d6ee3/AITracker/src/model.cpp#L11 "Update candidate: Tracker class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

AIRLegend, 2020-08-26:
I don't think it's worth the effort on that class. The only public member which is worth of checking not `nullptr` is `PositionSolver`. Maybe a shared_ptr or a getter/setter for updating it and rising an exception in case of wrong set. However, I think it's better to delegate that on the caller (aka. TrackerWrapper and TrackerFactory).

rbroker, 2020-08-26:
I don't mind taking a look at the client part.

closed 2020-08-27 by AIRLegend

elfring, 2020-08-28:
How do you think about to integrate any evolving software development approaches?

AIRLegend, 2020-08-28:
What do you specifically mean? @elfring

elfring, 2020-08-28:
@rbroker published a software development approach which looks promising.