AIRLegend/aitrack #30: Activating 'use remote opentrack client', this can never be deactivated again.

philstopford, 2020-08-27:
1) Fresh extraction of AITrack package.
2) Run software.
3) Check the 'use remote opentrack client' checkbox.
4) Restart tool.
5) Checkbox (3) is selected. Deselect.
6) Restart tool
7) Checkbox (3) is selected.

Checkbox is also checked after unchecking if the user presses 'apply'

AIRLegend, 2020-08-27:
If you have any text in the fields it assumes you're overriding the default configuration, so it checks the box. Simply clear the text, deactivate the field and click apply.

philstopford, 2020-08-27:
I've tried that :

Clearing the text in both boxes and hitting apply, it restores the text in both fields.
Clearing the text in both fields and unchecking the box, then clicking apply, it restores the text in both fields and checks the box.
Clearing the text in both fields and quitting the tool, on restart, the text is present in the fields and the checkbox is set.
Clearing the text in both fields and unsetting the checkbox, then restarting the tool, text is present in both fields and the checkbox is set.

AIRLegend, 2020-08-27:
Okay, managed to reproduce it. Thanks!

AIRLegend, 2020-08-29:
Already fixed in #34. Thanks for reporting it!

Closing.

closed 2020-08-29 by AIRLegend