AIRLegend/aitrack #32: For the Wiki, reduced latency with higher FPS on Logitech c920

pixelbreaker, 2020-08-28:
To get the best performance with the logitech c920 (and any webcam) it reduces lag greatly if you can switch the camera from 30fps to 60fps. To do this with the Logitech c920 the only way is to install [Logitech Capture](https://www.logitech.com/en-gb/product/capture) and change the camera to 360p and 60fps (see screenshot).

[image]

pixelbreaker, 2020-08-28:
I have tested this in FS 2020. In Dirt Rally 2 and Project Cars 2 this makes the head tracking much more useful.

AIRLegend, 2020-08-28:
Have you also tested leaving the camera with the default settings and editing `prefs.ini` in order to change the FPS? I could have the same effect.

pixelbreaker, 2020-08-28:
Hey, thanks for this. I just tried with 0.5.1 alpha and set my logitech capture settings back to defaults, then change my cam to 400x300 and 60fps, then 120fps, and it's amazing, Just did a quick flight in FS 2020, it's very very responsive now.

Thanks for your work on this!

AIRLegend, 2020-08-28:
Glad you like it!

Closing issue.

closed 2020-08-28 by AIRLegend