AIRLegend/aitrack #35: Crash after tracking start

meznak, 2020-08-30:
**Describe the bug**
Crash when starting tracking

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh copy of v0.6
2. Click on 'Configuration'
3. Select camera from dropdown
4. Click 'Apply'
5. Click 'Start Tracking'
6. Look around a bit
7. App crashes with no message

**Expected behavior**
App tracks properly without crashing

**Environment (please complete the following information):**
 - Win 10 Home x64 v1909 build 18363.1016
 - Webcam model: Logitech c260 (also active: LarmTek 1080p full HD)
 - AITrack version: 0.6

**Additional context**
Using with OpenTrack 2.3.12

AIRLegend, 2020-08-30:
Did you have the same issue while using 0.5.1?

n9533, 2020-09-01:
I'm having the same issue with 0.6 and 0.5.1

AIRLegend, 2020-09-05:
I've just released version 0.6.1 which adds a logging system. It could help us in figuring out what's happening. 
Please, post the content of `log.txt` here.

AIRLegend, 2020-09-06:
As commented on Discord, manually setting FPS=30 for your camera in the cofiguration window seems to fix the issue.

egovice, 2020-09-07:
> As commented on Discord, manually setting FPS=30 for your camera in the cofiguration window seems to fix the issue.

I found any positive number fixed it in my case; although an FPS of 1 isn't exactly usuable

meznak, 2020-09-07:
Not crashing in 0.6.1 after manually setting FPS.

Chaya56, 2020-09-10:
+1, by default FPS is 0 and it make crash app, setting it to another value make it work

AIRLegend, 2020-09-11:
Glad to hear that! The fix will be included with the new release. 
Closing.

closed 2020-09-11 by AIRLegend