AIRLegend/aitrack #36: Changing resolution not reflected in preview

meznak, 2020-08-30:
**Describe the bug**
When changing resolution to match the camera, the preview is not scaled, causing only a portion of the image to display. Tracking appears to still work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configuration'
2. Change resolution to match camera
3. Click 'Apply'
4. Click 'Start Tracking'
5. See a small portion of your face

**Expected behavior**
Preview should show full frame at new resolution

**Screenshots**
 - 1080p camera set to 640x480
[image]
 - same camera set to 1920x1080
[image]

**Environment (please complete the following information):**

 - Win 10 Home x64 v1909 build 18363.1016
 - Webcam models: Logitech c260 (720p), LarmTek 1080p full HD
 - AITrack version: 0.6

**Additional context**
Using with OpenTrack 2.3.12

AIRLegend, 2020-08-30:
I think it's not very user intuitive. I will have to tackle that.

Internally, AITrack gets video from your camera at its native resolution (in your case, 1920x1080) and then it's downscaled to the resolution you have in your preferences. This is to ensure no distortion and no crashing in the capture phase in case the user inputs some incompatible resolution.

When you set the 1080p resolution, you're simply "doing no downscaling", and the viewport shows  a cropped version of your feed (as the window size is fixed, the viewport resolution is 640x480), although AITrack is using the full scale video feed, so that's why the tracking still works.

isaacrlevin, 2020-09-02:
To add on to this, seems the camera preview, regardless of resolution does not match where the camera is pointing. For instance my face is centered in the camera app but is off here

[image]

There is a ton of black space underneath

AIRLegend, 2020-09-05:
This should be fixed with #45. Please, check with version 0.6.1 to see if it still happens.

meznak, 2020-09-07:
Not quite working here. Whole frame is displayed, but multiplied horizontally, breaking tracking.

[image]

AIRLegend, 2020-09-07:
That's another bug, probably produced by OpenCV. I'm not sure why it happens yet. However, lowering the resolution (if your camera allows it) seems to work as a temporary fix. 

Closing this issue as it is duplicated. I will update on my findings on #47

PS: I've already fixed having to manually input 30 FPS. Will be out with the next release.

closed 2020-09-07 by AIRLegend

Vajikovajiko, 2023-10-10:
hello can u help me?
When I am changing resolution from 1280x720 to 640x480 aitrack stops responding. What's the reason?

[2023-10-10 13:45:06.129] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2023-10-10 13:45:06.131] [aitrack] [info] Created/Found prefs.ini
[2023-10-10 13:45:06.477] [aitrack] [info] Searching for cameras...
[2023-10-10 13:45:08.071] [aitrack] [info] Number of recognized cameras: 2
[2023-10-10 13:45:08.074] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2023-10-10 13:45:08.074] [aitrack] [info] Building Tracker with selected camera: 0
[2023-10-10 13:45:08.124] [aitrack] [info] Tracker initialized.
[2023-10-10 13:45:08.124] [aitrack] [info] Updated stabilizer.
[2023-10-10 13:45:08.124] [aitrack] [info] Updating camera parameters...
[2023-10-10 13:45:08.124] [aitrack] [info] Updated camera parameters. 1280x720@30
[2023-10-10 13:45:08.124] [aitrack] [info] Checking for updates
[2023-10-10 13:45:08.170] [aitrack] [info] App initialized
[2023-10-10 13:47:30.130] [aitrack] [info] Saving prefs
[2023-10-10 13:47:30.130] [aitrack] [info] Updated stabilizer.
[2023-10-10 13:47:30.130] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2023-10-10 13:47:30.130] [aitrack] [info] Updating camera parameters...
[2023-10-10 13:47:30.130] [aitrack] [info] Updated camera parameters. 1280x720@30
[2023-10-10 13:47:30.130] [aitrack] [info] Tracker initialized.
[2023-10-10 13:47:30.130] [aitrack] [info] Prefs saved
[2023-10-10 13:47:34.778] [aitrack] [info] Saving prefs
[2023-10-10 13:47:34.778] [aitrack] [info] Updated stabilizer.
[2023-10-10 13:47:34.778] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2023-10-10 13:47:34.778] [aitrack] [info] Updating camera parameters...
[2023-10-10 13:47:34.778] [aitrack] [info] Updated camera parameters. 1280x720@30
[2023-10-10 13:47:34.778] [aitrack] [info] Tracker initialized.
[2023-10-10 13:47:34.778] [aitrack] [info] Prefs saved
[2023-10-10 13:47:37.578] [aitrack] [info] Saving prefs
[2023-10-10 13:47:37.578] [aitrack] [info] Updated stabilizer.
[2023-10-10 13:47:37.578] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2023-10-10 13:47:37.578] [aitrack] [info] Updating camera parameters...
[2023-10-10 13:47:37.578] [aitrack] [info] Updated camera parameters. 1280x720@30
[2023-10-10 13:47:37.578] [aitrack] [info] Tracker initialized.
[2023-10-10 13:47:37.578] [aitrack] [info] Prefs saved
[2023-10-10 13:47:40.978] [aitrack] [info] Saving prefs
[2023-10-10 13:47:40.978] [aitrack] [info] Updated stabilizer.
[2023-10-10 13:47:40.978] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2023-10-10 13:47:40.978] [aitrack] [info] Updating camera parameters...
[2023-10-10 13:47:40.978] [aitrack] [info] Updated camera parameters. 1280x720@30
[2023-10-10 13:47:40.978] [aitrack] [info] Tracker initialized.
[2023-10-10 13:47:40.978] [aitrack] [info] Prefs saved
[2023-10-10 13:47:42.416] [aitrack] [info] Starting camera 0 capture

here is log file