AIRLegend/aitrack #38: Sharing AITrack over Scoop

udit, 2020-09-02:
**Describe your idea**
Currently, you need to keep a watch on latest releases and download and installation of AITrack is manual. It would be great if AITrack is added to [Scoop](https://scoop.sh/).

Wiki : https://github.com/lukesampson/scoop/wiki

AIRLegend, 2020-09-02:
Certainly an update notifier is a must, but I don't think the CMD is the best way as only developers use these kind of tools. I think it would be better to build a feature to check whether is a release on Github more recent than the current running version and then, notify the user.

udit, 2020-09-03:
In my opinion, it could be both. People like me who use Scoop would benefit from the convenience of it. [OpenTrack](https://github.com/lukesampson/scoop-extras/blob/master/bucket/opentrack.json) is also onboard.

closed 2020-09-11 by AIRLegend

udit, 2020-09-13:
I think we need to get the json added to the extras bucket for this to work: https://github.com/lukesampson/scoop-extras/tree/master/bucket

AIRLegend, 2020-09-13:
You can already add this repository as a bucket:

`scoop bucket add https://github.com/AIRLegend/aitrack.git`

I don't know, however, if this app qualifies for being added there right now because of its "testing stage".

Try opening an issue / PR there to see if it can be added :)