AIRLegend/aitrack #39: Crash

n9533, 2020-09-02:
**Describe the bug**
Crashes when I press "Start". I have tried this on 0.6, 0.5.1 and 0.5. I have to re download the app every time I use it. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start app
2. Click on 'Start'
3. *Crashes with no report

_It does the same thing when I run as Admin_

**Expected behavior**
Ability to use the app, close it and use it again with no issues of crashing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version 10.0.19041 Build 19041
)
 - Webcam model: EpocCam iPhone Xr
 - AITrack version: 0.6, 0.5.1, 0.5

**Additional context**
Add any other context about the problem here.

AIRLegend, 2020-09-03:
Check this, https://github.com/AIRLegend/aitrack/issues/27#issuecomment-683853109

Also, try with Droid Cam. I don't think that's the problem, but just to be sure...

AIRLegend, 2020-09-05:
I've just released version 0.6.1 which adds a logger. It should help us to discover what's happening. 
Please, post the content of `log.txt` here.

vanvancool, 2020-09-06:
It's crashed with version 0.6 and 0.61 when i hit start button
Preview not working, camera led is activated for about 1 sec. (during app initializing)

But works fine with version 0.51

Tested with my very old Webcam (about 2008): Logitech QuickCam Pro 9000 

Here crash log 0.61 ( hit "start" button ):

[2020-09-06 12:55:46.861] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-06 12:55:47.305] [aitrack] [info] Created/Found prefs.ini
[2020-09-06 12:55:47.305] [aitrack] [info] Searching for cameras...
[2020-09-06 12:55:49.255] [aitrack] [info] Number of recognized cameras: 1
[2020-09-06 12:55:49.256] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-09-06 12:55:49.256] [aitrack] [info] Building Tracker with selected camera: 0
[2020-09-06 12:55:49.324] [aitrack] [info] Tracker initialized.
[2020-09-06 12:55:49.324] [aitrack] [info] Updated stabilizer.
[2020-09-06 12:55:49.324] [aitrack] [info] Updating camera parameters...
[2020-09-06 12:55:49.324] [aitrack] [info] Updated camera parameters. 640x480@0
[2020-09-06 12:55:49.325] [aitrack] [info] App initialized
[2020-09-06 12:55:50.109] [aitrack] [info] Starting camera 0 capture
[2020-09-06 12:55:50.734] [aitrack] [info] Camera 0 started capturing

AIRLegend, 2020-09-06:
@vanvancool As commented on Discord, manually setting FPS=30 for your camera in the cofiguration window seems to fix the issue. (Issue #35).

vanvancool, 2020-09-06:
confirmed, thank you for the incredibly fast answer

gb056, 2020-09-08:
Can also confirm this fixed the exact same issue for me.

AIRLegend, 2020-09-11:
Seems the same bug as #35, so I will close this issue just to clean up a little bit.
If you find the problem still persists, please, open another one. Thanks for reporting :)

Closing.

closed 2020-09-11 by AIRLegend