AIRLegend/aitrack #40: Crash at start

Deci781, 2020-09-03:
**Describe the bug**
Crash after launch the app

**To Reproduce**
Steps to reproduce the behavior:
1. Launche the app

**Expected behavior**
Ability to use the app, close it and use it again with no issues of crashing.

**Screenshots**
1s after launch can't screen

**Environment (please complete the following information):**
 - Operating system version Windows 10 last version
 - Webcam model: Logitech Quickcam IM/Connect
 - AITrack version: 0.6

**Additional context**
It does the same thing when I run as Admin

AIRLegend, 2020-09-03:
Hi!
Check this out, https://github.com/AIRLegend/aitrack/issues/27#issuecomment-683853109
Also, make sure you're running *AITrack.exe* next to the `models/` folder.

Hope it helps.

Deci781, 2020-09-04:
> Hi!
> Check this out, [#27 (comment)](https://github.com/AIRLegend/aitrack/issues/27#issuecomment-683853109)
> Also, make sure you're running _AITrack.exe_ next to the `models/` folder.
> 
> Hope it helps.

Hello, i read it, but everything is fine on this way. 
Still crashing

AIRLegend, 2020-09-05:
Hi! version 0.6.1 is out, which adds a logger. It should help us to discover what's happening. 
Please, post the content of `log.txt` here.

Deci781, 2020-09-06:
Thanks for support, 
[2020-09-06 13:28:27.114] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-06 13:28:27.963] [aitrack] [info] Created/Found prefs.ini
[2020-09-06 13:28:27.963] [aitrack] [info] Searching for cameras...

If it don't find a camera, it will close ? Or it's really a crash ?

AIRLegend, 2020-09-06:
It seems to be while searching for available cameras. Happens the same if you try using Droid Cam instead of that webcam?

Deci781, 2020-09-06:
> It seems to be while searching for available cameras. Happens the same if you try using Droid Cam instead of that webcam?

Hello, its the same. Immediate crash.

AIRLegend, 2020-09-11:
I think an antivirus or windows itself could be flagging AITrack as malware and preventing it from accessing the camera. Check it has camera permissions (via windows privacy menu), that your AV isn't blocking it and neither other app.

PS: Sorry for the response delay.

Deci781, 2020-09-14:
> I think an antivirus or windows itself could be flagging AITrack as malware and preventing it from accessing the camera. Check it has camera permissions (via windows privacy menu), that your AV isn't blocking it and neither other app.
> 
> PS: Sorry for the response delay.

Hello, i tried to stop my AV and nothing happened. I have already check privacy menu for camera. To be continued ... ^^

SpaceEnthusiast, 2020-09-14:
> Hello, i tried to stop my AV and nothing happened. I have already check privacy menu for camera. To be continued ... ^^

I had the same issue and it was solved when I added AITrack.exe to exclusions in Windows Defender. Have you tried that?

closed 2021-04-18 by AIRLegend