AIRLegend/aitrack #46: Run OpenTrack remotely (e.g. on a laptop w/ webcam) while running AITrack locally (e.g. on a desktop w/o webcam)

GamerKingFaiz, 2020-09-05:
**Describe your idea**
I'd to be able to run OpenTrack remotely on my laptop that has a webcam. Then I'd like AITrack to be able to hook into the OpenTrack feed.

**Describe the solution you'd like**
It looks like network capabilities are already built into both (you have to specify an IP address and port), but I'm not sure if there are any other limiting factors.

**Additional context**
Currently when I open AITrack on my desktop it will error out saying that there's no webcam connected. And the options are greyed out to input my laptop's IP address (can't even tick the checkbox).
[image]

AIRLegend, 2020-09-05:
You should open AITrack on your laptop and then put the set the "Use remote Opentrack" fields with the machine running Opentrack (your desktop in this case). AITrack will then send the data via UDP to your main computer.

GamerKingFaiz, 2020-09-05:
Oh yeah, I guess I had the order backwards. That did it--thanks!

closed 2020-09-05 by GamerKingFaiz

pedro-1952, 2021-06-10:
Hey I am having problems setting the port number. Which port did you use?

GamerKingFaiz, 2021-06-10:
I don't remember because it's been awhile and I ended up using a local webcam. But I'm pretty sure I just used the opentrack default, which seems to be `4242`. Make sure you're using the same port in AITrack too.

Otherwise make sure it's not a firewall issue or some other reason the port isn't working. Maybe post screenshots of your settings and any errors you get if you still can't get it work.