AIRLegend/aitrack #48: Camera starts and immediately stops

AF1E, 2020-09-09:
When I start the app the camera starts and stops.  I see light on camera come on and then immediately quit.  Flight Sim seems to be connected for that second.  

Log data
[2020-09-09 15:26:34.535] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-09 15:26:34.831] [aitrack] [info] Created/Found prefs.ini
[2020-09-09 15:26:34.831] [aitrack] [info] Searching for cameras...
[2020-09-09 15:26:35.645] [aitrack] [info] Number of recognized cameras: 1
[2020-09-09 15:26:35.646] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-09-09 15:26:35.646] [aitrack] [info] Building Tracker with selected camera: 0
[2020-09-09 15:26:35.687] [aitrack] [info] Tracker initialized.
[2020-09-09 15:26:35.687] [aitrack] [info] Updated stabilizer.
[2020-09-09 15:26:35.687] [aitrack] [info] Updating camera parameters...
[2020-09-09 15:26:35.687] [aitrack] [info] Updated camera parameters. 640x480@0
[2020-09-09 15:26:35.687] [aitrack] [info] App initialized
[2020-09-09 15:26:38.885] [aitrack] [info] Starting camera 0 capture
[2020-09-09 15:26:39.119] [aitrack] [info] Camera 0 started capturing

Running Win 10 2004 and is up to date with patches
Microsoft Flight Sim 2020
Camera Logitech Quickcam Ultravision
[image]

AIRLegend, 2020-09-09:
Hello @AF1E. That's a known bug. The current fix is to manually enter 30FPS on the configuration window.
Please, report if that solves your problem.

AF1E, 2020-09-09:
That fixed it.  Need to do some more testing.  Thanks.

AIRLegend, 2020-09-10:
Great! Closing.

closed 2020-09-10 by AIRLegend