AIRLegend/aitrack #49: Roll is glitching

LiftPizzas, 2020-09-09:
Roll value quickly swaps from 0 to 180. It happens more often when I'm looking straight forward.

**To Reproduce**
I'm not sure if it will reproduce on anyone else's setup. Linked is a video showing the result at the beginning and then all of the data glitching.

**Expected behavior**
Quick/huge changes should be ignored and not considered valid input/output.

**Screenshots**
https://youtu.be/mDlfIRWKJgQ

**Environment (please complete the following information):**
 - Windows 10 Home (v2004) build 19041.450
 - Webcam model: Logitech C920
 - AITrack version: 0.6.1-alpha

**Additional context**
Output Roll is also centered at 90. AFAIK it should be centered at 0.

(Also huge thanks for the software so far, it's awesome!)

AIRLegend, 2020-09-10:
Hello! Thanks for the video! Very helpful.

Roll being centered at 90 should not matter. It's range is [0, 180], so 90 is half the way.
Judging by how all the values jump (X, Y and Z also go crazy): Have you tried different values of the Distance parameter? Also with/without "Landmark stabilization" enabled?

And this is off topic from your problem, but the C920 seems to be a pretty conflictive camera: Have you noticed low FPS on the video feed? Also, does the video look right if you set the capture resolution to 1920x1080?

LiftPizzas, 2020-09-10:
Sorry in advance for the verbose bug report. ;-) I had tried distances and landmarks on/off, also different FPS values and all the other settings. I film let's plays with the same webcam so I have a good lighting setup as well. With anything other than 640 x 480 I am getting the "images side by side" thing and it twitches between the multiple images (the pink markers are changing their mind about which of the three faces they belong to.)

I think some of the issues with Z are because the webcam insists on defaulting to auto-focus, and I would  expect it to lose track when the focus changes.

I also hooked up a QuickCam S7500. It is a lot better, but also has a place where it goes twitchy.

IMO it would help to have a minimum time/value threshold, any change above this amount in less than this much time gets ignored. (You can only move your head so fast unless you're being punched in the jaw. :) It would add some latency to output depending on your time factor, and a few more things to the settings, and the need to store a few frames of past data. But I don't think it's going to be possible to ensure the camera never loses track, and that twitching is going to be a big problem for some users. (I can tolerate it but the 3 or 4 people who watch my videos probably won't.)

karoletrych, 2020-09-22:
Got similar issue with all of the values glitching on Logitech C270 (which doesn't have autofocus): 
https://www.youtube.com/watch?v=k6IYNwVJtt8
https://www.youtube.com/watch?v=jvwk6xHQHmM

AIRLegend, 2020-10-16:
Sorry about the delay, but v0.6.3 seems to have solved this. However, there's one bug: you won't be able to update the FOV value using the options menu. If you need to change it, please, edit `prefs.ini` (for the moment).

fkrauthan, 2020-10-23:
Maybe I screwed up something on the setup since I never used opentrack but the tracking preview in AiTrack looks good (all the tracking points are where the belong regardless of how I move my head) but opentrack preview shows when I move my head left and right that I also move up and down (and in some cases inverts me like some of the other people posted here). In general even if I move my head only in a single axes I always seem to move on multiple on the opentrack preview (and in game).

AIRLegend, 2020-10-23:
Hi @fkrauthan. Have you tried with Medium/Heavy trackers?

fkrauthan, 2020-10-23:
Yes I played around with Medium, Heavy and disabling the stabilization. The AiTrack preview shows a perfect track (no visible flashing or anything). But the movement in OpenTrack just does not match up to that. Even where opentrack flips the image (if I look to far left or right) the track itself still looks like its fully working.

EvilMav, 2021-05-16:
Same for me. Roll tends to jump to -180 when looking down or to the side on a MS Lifecam Cinema, even though the tracking looks fine preview. Tried switching to heavy etc.

searching46dof, 2022-07-01:
this may be related to the issue i reported regarding jitter "need wider range of model type to reduce jitter #145"
I observed large amounts of jitter without moving my head at a distance of 4meter. In the preview window I can see the rectangle and facial detection pink dots move alot.

the fix to the boundary conditions in the function logit seems to help significantly and the optimizations to MAFilter may be contributing as well.