AIRLegend/aitrack #54: High framerates implies poor brightness and so poor tracking (?)

ecidou, 2020-09-16:
(First of all, a big thank you for your software !)

**Describe the bug**
I noticed poor tracking even after understanding the Wiki.
An i think i found a clue = it depends on the framerate (?!)

**To Reproduce**
Steps to reproduce the behavior:
1. CL-Eye Test : gain=min, exposure=max, white balance=max  
    - The brigthness is **very good in the CL-Eye Test**
2. Now in aitrack, of course i use the best FPS for the PS3 Eye : **75**  (with **no** custom brightness)
    - The brigthness is **very poor in aitrack**
3. In aitrack, I **decreased the FPS** : at 10 
    - The brigthness is **as expected in aitrack** BUT poor tracking smoothness due to the 10 FPS ... 
    - Acceptable brighness is ~30 ... which is far below the PS3 eye's capabilities ... :-(
4. I tried the custom brightness in aitrack 
    - The brigthness is **even worse in aitrack** 

(see the screenshots below)

**Expected behavior**
The same brightness as the CL-Eye Test to allow a good tracking

**Screenshots**
[Screenshots which show clearly the issue](https://1drv.ms/u/s!Am_-yrC4_SQ_y0QLhW8KlPjRcS3z?e=uEpMcc)

**Environment :**
 - Operating system version (Windows 10 - 2004): 
 - Webcam model: PS3 Eye (driver 5.3.0.341)
 - AITrack version: 0.6.2-alpha

ecidou, 2020-09-19:
Hi, this morning i tried to find clues about this issue, and I play a little with **Opentrack** (not aitrack) and even if I've no caps or clips, i  use the **Point tracker** to see my face. And to check: 
1. If the brigthness is not too dark (like aitrack) with the **same driver** ?
2. If the brigthness custom sliders in Opentrack works ? (because in aitrack they do not increased the brighness, even pushed to the right)

The answers ... for the both is **everything is ok with Opentrack** ! Even with low light.
So it is definitely an aitrack bug ? But a "strange" hope appears: I execute again aitrack and now the brightness custom sliders work like a charm !?!? And so the 75 HZ tracking is very very good !

So my understanding is now : before i use OpenTrack with the **Point tracker**, it was like: 
1. aitrack does not use the brighness config set in CL-Eye Test program (~ previous post)
2. aitrack custom sliders were **locked** somewhere (= don't act on the brightness driver/camera config), and it seems that playing with the Point tracker in Opentrack unlocked them !?!? Don't ask me why :-)
3. But sometimes i noticed that aitrack doesn't kept ticked the custom brightness box

I reboot to test, and it's  still ok

Moreover, i found a clear explanation about the "brigthness depending on the FPS", which is not a bug :
- Page 7, "Frame Rate (FPS)" [here](https://steamcdn-a.akamaihd.net/steam/apps/929270/manuals/ViewTracker_manual.pdf?t=1565220890)

Hope it helps to fix something (or to explain in the wiki)
Regards

AIRLegend, 2020-09-20:
Thanks for your digging efforts :)

The camera FPS will affect the brightness it perceives (the more FPS, the less light the camera will be able to capture per frame and thus, the darker the image). So, as you pointed, if you want to use higher FPS, you should have your face more lit. Similarly, increasing the exposure, should decrease the real FPS. That's a camera/driver thing, not part of AITrack's behaviour.

As for the sliders, right now (v0.6.2), they only work on PS3 cameras (the only one I have for testing right now). Its values, after clicking "Apply", should always override the ones set in the CLEye driver. 

Also, could you please tell me the steps to replicate the behaviour of the custom brightness checkbox not being saved? I'm not been able to replicate it.

ecidou, 2020-09-20:
For your 1st point : yes, now i understand, but the screenshoots facts show (with the **same sliders values**)  that aitrack get **less** light than CL-Eye Test or OpenTrack (with Point tracker)
For your 2nd Point : I have this PS3 Eye camera. And no, after cliking on Apply, if you exit from aitrack, and execute CL-Eye Test, the driver values is not set as in aitrack. But if you do it in OpenTrack, the driver value are the same when you execute CL-Eye Test. So aitrack kept its own value, but do not set in the driver.
For your last point, it is unfortunalty random :( (if i find a clue i'll keep you informed)