AIRLegend/aitrack #55: My wife want to see your software ... and aitrack don't know who to look at !

ecidou, 2020-09-16:
**Describe the bug**
Two heads in the FOV of the camera and the tracking become erractic

**To Reproduce**
Just add another person in front of the camera.
Sometimes you see the camera passing from one face to the other, so my plane crashed :-)

**Expected behavior**
Just the more "centered" face should be tracked (?)

searching46dof, 2022-06-27:
I was flying when the face tracking was jumping back and forth. my son was sleeping on a sofa in the background.

a possibly easier way to implement this may be to limit the facial detection within a centered rectanglular window of configurable size (a digital zoom). if the position of the rectangular windows was also configurable instead of being centered, it may be more useful since the camera would not need to be manually reposititoned.

AIRLegend, 2022-09-09:
Sorry for the delay, but @searching46dof added this feature and it's available on 0.7.0 :)

Closing the issue