AIRLegend/aitrack #58: Aitrack crashing when using PS3 EYE

ProrokC2, 2020-09-25:
**Describe the bug**
When I try to use my PS3 eye camera, it either crashes instantly, or works for a while and then crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Open AITrack
2. Select PS3Eye (for me cam1)
3. Press Start Tracking
The tracking will either work, or freeze instantly.
While the tracking is working, it will ocasionally freeze, for a few seconds, and come back to life. It will continue freezing untill one freeze causes a crash.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number):  win 10 pro
 - Webcam model: ps3 eye
 - AITrack version: 0.6.2

**Additional context**
Logs : 
[2020-09-26 01:16:07.578] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-26 01:16:07.974] [aitrack] [info] Created/Found prefs.ini
[2020-09-26 01:16:07.974] [aitrack] [info] Searching for cameras...
[2020-09-26 01:16:08.722] [aitrack] [info] Number of recognized cameras: 2
[2020-09-26 01:16:08.725] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-09-26 01:16:08.725] [aitrack] [info] Building Tracker with selected camera: 0
[2020-09-26 01:16:08.783] [aitrack] [info] Tracker initialized.
[2020-09-26 01:16:08.783] [aitrack] [info] Updated stabilizer.
[2020-09-26 01:16:08.783] [aitrack] [info] Updating camera parameters...
[2020-09-26 01:16:08.783] [aitrack] [info] Updated camera parameters. 640x480@60
[2020-09-26 01:16:08.783] [aitrack] [info] Checking for updates
[2020-09-26 01:16:08.808] [aitrack] [info] App initialized
[2020-09-26 01:16:10.180] [aitrack] [info] Starting camera 0 capture
[2020-09-26 01:16:10.193] [aitrack] [info] Camera 0 started capturing

AIRLegend, 2020-09-26:
Are you using your camera via USB Hub? Does it stil happen if you connect your camera into another port?

ProrokC2, 2020-09-28:
So after countless hours i fixed the issue. It was a faulty(?) usb port (i had the joystick and camera on front case usb 2.0 ports, maybe the bus had too little badwidth or something) and driver issues. Now it works fine.

closed 2020-09-28 by ProrokC2