AIRLegend/aitrack #59: Instant crash: Searching for cameras

ToniVerch, 2020-09-26:
After Executing AITrack.exe the application crashes within 3 Seconds. I cant see a GUI.
Windows 10, generic HD Camera (60Hz)

Already tried:
- priviliged execution
- windows firewall execution
- Version 0.6.2 & 0.6.1

log.txt contains:
[2020-09-25 21:06:39.323] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-09-25 21:06:39.797] [aitrack] [info] Created/Found prefs.ini
[2020-09-25 21:06:39.797] [aitrack] [info] Searching for cameras...

AIRLegend, 2020-09-26:
Hi!
Check neither windows privacy settings nor antivirus are blocking AITrack.exe

ToniVerch, 2020-09-26:
Cant see AITrack within privacy settings, but I allowed the use of my camera for external apps. I have no antivirus installed.

closed 2021-04-18 by AIRLegend