AIRLegend/aitrack #60: Application crash

kinou777, 2020-10-03:
**Describe the bug**
Application does not respond / crashed, in to 2 cases : 
- At launch
- When starting the tracking

**To Reproduce**
Aitrack is crashing one time over 2
When successfull started, it is crashing everytime when launching the tracking
In any cases, application is never working as expected.

**Screenshots**
2 logs available :

Crash at start : 
[2020-10-03 12:41:07.032] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-10-03 12:41:07.265] [aitrack] [info] Created/Found prefs.ini
[2020-10-03 12:41:07.265] [aitrack] [info] Searching for cameras...

Crash at tracking start :
[2020-10-03 12:45:21.465] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-10-03 12:45:21.714] [aitrack] [info] Created/Found prefs.ini
[2020-10-03 12:45:21.714] [aitrack] [info] Searching for cameras...
[2020-10-03 12:45:23.378] [aitrack] [info] Number of recognized cameras: 2
[2020-10-03 12:45:23.379] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2020-10-03 12:45:23.379] [aitrack] [info] Building Tracker with selected camera: 0
[2020-10-03 12:45:23.420] [aitrack] [info] Tracker initialized.
[2020-10-03 12:45:23.420] [aitrack] [info] Updated stabilizer.
[2020-10-03 12:45:23.420] [aitrack] [info] Updating camera parameters...
[2020-10-03 12:45:23.420] [aitrack] [info] Updated camera parameters. 1280x720@30
[2020-10-03 12:45:23.420] [aitrack] [info] Checking for updates
[2020-10-03 12:45:23.457] [aitrack] [info] App initialized
[2020-10-03 12:45:29.839] [aitrack] [info] Starting camera 0 capture

**Environment (please complete the following information):**
 - Windows 10 Professionnal,  Version 19041.508 (latest) 
 - Webcam model: Logitech C920 Pro
 - AITrack version: 0.6.2 alpha

**Additional context**
Resolution method tried :
- Stopping AV
- Stopping Windows Defender
- Checked privacy settings for Camera and AiTrack
- Disabling/unistalling any third party software using the webcam (including logitech's one)
- Changing all AiTrack options (camera supported resolution and FPS, model, etc....)
- Granting administrator rights to AiTrack
- Deleting the webcam from windows device manager
- Changing USB port for Webcam

AIRLegend, 2020-10-03:
Hello! Thanks for the info!

Seems to be the same bug some people are experiencing (some other issues are already posted here). I don't think it's a hardware issue as there are quite a few of people who are using that camera. However, until now, I've been unable to get to the root of the problem. My guess for the moment is that windows could be interfering in some way with OpenCV while accessing the camera device. That would explain both types of crash you described.

Quick questions: 
- Does the crashing behaviour change at random or did you do something in order to get to the 2nd step?
- Do you have only one camera available under Options>Camera? If not, try with others.
- In the second case, does it crash if you set other resolution? (1080p, for example).

kinou777, 2020-10-03:
Hello ! Thanks for quick reply !
- The crash seems random, usually after 1 or 2 (max) crashes at start, the application is starting as usual.
I confirm I do not do anything in between my attempts.
- I don't have a second camera. I had previously a software (Snap Camera) installed emulating a 2nd device that was available in the AiTrack camera menu. I removed it to troubleshoot.
- I tried different resolutions supported by the camera (1920x1280, or even 640x480, all with 30 FPS), with no different behaviour that I could notice....
Hope this is helping.

AIRLegend, 2020-10-03:
Do you get the same error if you use Snap Camera (or any other virtual camera)?

Thanks!

kinou777, 2020-10-04:
Indeed working with snap Camera selected.

AIRLegend, 2020-10-04:
Strange then... Do you have your camera connected via USB hub?

kinou777, 2020-10-04:
No, directly to pc, I've tried different ports, make no difference.

closed 2021-04-29 by AIRLegend