AIRLegend/aitrack #61: Crash when searching for a cam.

Chewbacca626, 2020-10-06:
I already tried disabling webroot my antivirus and I also tried without windows defender, I am using this cam off of amazon (https://www.amazon.com/gp/product/B08FMCZCZY?pf_rd_r=3ST4YWE0NNZ3V4BFTGGH&pf_rd_p=edaba0ee-c2fe-4124-9f5d-b31d6b1bfbee&th=1) and the log is below

[2020-10-05 20:28:33.939] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2020-10-05 20:28:34.167] [aitrack] [info] Created/Found prefs.ini
[2020-10-05 20:28:34.167] [aitrack] [info] Searching for cameras..

AIRLegend, 2020-10-06:
Hi!
That's an already reported bug, please, see #40, #59 or #60. Maybe something proposed there will work with you. In case not, know that I'm already trying to fix that :).

I'm going to close this issue in order to avoid too many duplicates. Will report when I find the solution.

Thanks!

closed 2020-10-06 by AIRLegend

Chewbacca626, 2020-10-06:
Thanks man I tried most of the other stuff I saw and yeah I’m glad your
working hard, keep up the great work!!