AIRLegend/aitrack #62: Code Reference

xuhao1, 2020-10-16:
@AIRLegend 
Thanks for your outstanding contribution of facial head tracker for gaming.
I am also developing an open-source non-commercial facial head tracker [FlightAgentX](https://github.com/xuhao1/FlightAgentX) which combines landmark and end-to-end methods for head pose estimation with Kalman filter. I found your project today at Bilibili, which performs excellent and your pre-train model from @emilianavt is much better than the one provided by dlib. So I would like to integrate the model to replace the dlib model. I used pieces of your code of few functions including proc_heatmaps, normalize and transpose and will include your LICENSE and a link point to your repo in my own repo in next few commits.
I am writing this just to let you know.

AIRLegend, 2020-10-16:
Cool! Glad you found the code useful :) Also, feel free to make a PR if you think you can improve anything.

xuhao1, 2020-10-16:
Thanks! I will put PR if I can improve this project!

emilianavt, 2020-10-16:
Hi! I am very happy that my tracking models can be useful for both projects!
Just please note that my OpenSeeFace project, which the models are part of, is also released under a 2-clause BSD license, which is required to be distributed alongside them. If you could just include it in the models folder, or where ever else is convenient, that would be great.

AIRLegend, 2020-10-16:
Oh, sorry @emilianavt, I forgot.  Already added :).

Thanks again for the ONNX models, they're very handy with C++!

closed 2020-10-16 by AIRLegend

emilianavt, 2020-10-16:
Thank you! I appreciate it!

I didn't just do the ONNX conversion but also built them from scratch, so seeing them being used is really nice. ☺️

xuhao1, 2020-10-17:
> Thank you! I appreciate it!
> 
> I didn't just do the ONNX conversion but also built them from scratch, so seeing them being used is really nice. ☺️

Thanks for your contribution! And I will include your license and your repo link correctly.
I also have a question, estimate facial landmarks from 224x224 image is not neccassary sometime since my input image is near 100x100, is that possible to rescale the network?
I will try this by my own also.

xuhao1, 2020-10-17:
> Oh, sorry @emilianavt, I forgot. Already added :).
> 
> Thanks again for the ONNX models, they're very handy with C++!
Onnx is really convenient but it have bugs on multithread. I recompile it without openmp to run correctly.
You may also see the onnx model under my project, I convert [FSANet](https://github.com/shamangary/FSA-Net) @shamangary to onnx model and use it for rotation estimation. The network is robust and really fast, takes only few millisecond to inference.

AIRLegend, 2020-10-17:
I think I already explored that NN during initial testing, but I found @emilianavt 's models to be more robust and lighter.
As for the multithreading, I'm only using one thread for the moment (with no noticeable issues) because it already runs faster than most of the cameras people are using (although reducing inference CPU consumption would always be great).

emilianavt, 2020-10-17:
I meantioned it over [here](https://github.com/emilianavt/OpenSeeFace/issues/9#issuecomment-710783790) for @xuhao1's use case. Since you are mainly interested in the head pose and not facial features, you could give the new 30 point model a try. It's noisier, but I find that with higher smoothing it can still give acceptable results and it runs at about 5x the frame rate as the slowest model for me.