AIRLegend/aitrack #65: Problems with camera position.

uchiki, 2020-10-16:
I can't put the camera completely in front of me and it's a little above me. (About 10 degrees)

In this case, if I turn left and right, the output in the pitch direction will be added against my intention.

AIRLegend, 2020-10-17:
I also have the camera in a similar position. It sounds like you have swapped axes in Opentrack (Opentrack>Options>Output).

uchiki, 2020-10-17:
Looking at the Raw tracker data in opentrack.

When I turn my head to the left or right, the yaw changes accordingly, but the pitch also changes.
(If I turn my head 30 degrees to the left and right reading the Raw tracker data, the pitch will change by 8 degrees.)

My problem is that when I turn sideways, I get unintended movement in the pitch direction.

AIRLegend, 2020-10-17:
Are you using the latest version (v0.6.3)? Happens the same with all the trackers? Have you tried different distance parameter values?

uchiki, 2020-10-17:
I am trying with v0.62 and v0.63. 

The camera distance is measured with a ruler. (0.8 meters)
I tried shifting the distance by 0.3 meters, but it didn't solve the problem.

This problem is less with FaceTrackNoIR. 
(The width of the raw input pitch when facing left and right is about 2.5 degrees)

There are other issues with this app, so I'm trying your app.