AIRLegend/aitrack #67: Minimize to Tray Option

SeriousOldMen, 2020-10-21:
Hi,

great little tool...

It would be great to have some startup options like Minimize to Tray and Start Tracking on startup. An option for Autostart with Windows would be nice as well, but this could be solved with the WIndows Autostart folder.

And, is there a way to donate to this project?

Best regards

Oliver

AIRLegend, 2020-10-23:
I can add an "Auto start tracking on launch" option without any problem! Thanks!

velotoxicsounds, 2021-01-04:
Is this feature already implemented?

AIRLegend, 2021-01-07:
Not ATM

erasam, 2021-12-04:
Minimize to try is a must have if you have a very crowded application bar

ninjax95, 2022-02-02:
> I can add an "Auto start tracking on launch" option without any problem! Thanks!

Hi, some news for this option ? Thanks