AIRLegend/aitrack #68: Tracking with headset on

Xebozone, 2020-10-22:
Thanks for all your hard work. I love the app!

**Describe your idea**
The app doesn't work too great with a headset on (Phone VR with Riftcat using Freetrack tracking)
I suppose it looks for the eyes which cannot be seen with a headset on.

**Describe the solution you'd like**
Rely just on head shape/mouth tracking, or add the ability for manual tracking points

**Describe alternatives you've considered**
I could stick some printed eyes on the headset, but that's creepy 🤣 

**Additional context**
Thanks!

Xebozone, 2020-10-22:
This looks cool which may be a useful library FYI:
https://github.com/google/mediapipe

Found it while reading this project:
https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr

AIRLegend, 2020-10-23:
Thanks for the idea! I actually tried MediaPipe's models but found that they  are extremely noisy for this (as they're intended for their use on mobile phones). 
On the other hand, support for VR headsets is not on the near roadmap, sorry!

closed 2020-10-23 by AIRLegend