AIRLegend/aitrack #69: Multiple Camera Support

Xebozone, 2020-10-22:
**Describe your idea**
The app works great for 1x camera, but I want to move around the room or maybe just turn my head around to face another camera.

**Describe the solution you'd like**
This is obviously a much larger undertaking.
The ability to add in multiple cameras to track the user would potentially allow for full-room, or at least allow the user to turn their head to face another camera.

Each camera would have to have an option to specify where it is in the 3D space (offset x,y,z from perhaps a main camera?)
It would be cool if this could be calculated/calibrated automatically by tracking the face and its size as it moves around, maybe just then add a zoom factor and the ability to override this (This would also obviously work better if using multiple of the same camera (same zoom, fov)).

AIRLegend, 2020-10-23:
I suppose you're talking about VR. In that case, wouldn't the target user already have some system for tracking himself? (like VIVE base stations).

Xebozone, 2020-10-25:
Thanks for your reply!
Yes, I was thinking of a "poor man's VR" solution by using spare webcams to track the user.
However having multi-camera support may also be useful in other instances, for example, if the user has monitors to the left and right of them with webcams mounted on top (think 270 degree surround-monitors instead of actual VR), then your app could continue tracking as they turn their head to face those other monitors.

AIRLegend, 2020-10-30:
I think this is a very good idea for VR body tracking solution rather than a simple head tracker for games (or at least, the standard use case of this program). It could be very useful :)