AIRLegend/aitrack #70: Error on windows 10

marcelcandidoferreira, 2020-10-24:
**Describe the bug**
I'm trying to open it in windows 10, but receiving the error bellow
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open droidcam client and connect to webcam
2. Open OpenTrackAi and select input UDP Over Network, then click in Start
3. Try to open AiTrack and get the error

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number):  Windows 10  Version	10.0.19041 Build 19041
 - Webcam model: DroidCam
 - AITrack version:  aitrack-v0.6.3-alpha

AIRLegend, 2020-10-25:
Are you executing an AITrack.exe  which is next to all the required QT dll's? (Included within the .zip)

peter74-sys, 2021-02-08:
problem here
aitrack.exe will not run on windows 10
downloaded v0.6.3 alpha and tried older ones
get blank page and then stop working

AIRLegend, 2021-04-29:
@peter74-sys Your problem should be solved on v0.6.4

Closing the issue because of inactivity.

closed 2021-04-29 by AIRLegend