AIRLegend/aitrack #73: Image goes upside down when using yaw

Yenikoko, 2020-10-26:
Hi,

In FS2020, the image goes upside down when i'm using the yaw to the left or right.  after a certain degree (close to end of the range) the image goes upside down

mote0230, 2020-10-28:
For me this mostly goes away by putting model type at medium or heavy in the options.

seanamosw, 2020-10-30:
Having the same problem. It happens only when I turn my face towards the right. Very visible in the Opentrack preview, the whole Octopus beings turning, but then rolls upside down.

Swapping back to 0.6.2, the problem dissapears.

AIRLegend, 2020-10-30:
Hi. Thanks for reporting.

I know where the problem lies, but I'm not sure how to fully solve it yet. It should be mitigated on 0.6.3 with Medium/Heavy trackers, as @mote0230  says, as they use more facial points to do the position calculation. However, seems odd @seanamosw's case because Fast tracker it's actually the same as the one in 0.6.2... 

Have you tried moving your camera to other position? Maybe that helps.

Will update as soon as I find something.

Thanks :)

1234filip, 2020-11-15:
> Having the same problem. It happens only when I turn my face towards the right. Very visible in the Opentrack preview, the Octopus begins turning, but then rolls upside down. The preview in AITrack looks fine.
> 
> Swapping back to 0.6.2, the problem dissapears.

+1 on this, reverting seems to help.

Yenikoko, 2020-11-26:
Hi, thanks for your responses, I've change the tracking to Heavy and it seems like it has eliminated 95% of the issues, playing with lightning and angle of camera, I'm able to get a pretty good output now.

gconry18, 2020-12-08:
Just a note here that I am experiencing this issue when using heavy in a well lit room. 

The following might help you investigate:
Seems to happen at around 40 degrees of yaw and in my case, mostly only when turning my head left. 
The problem seems to be less prevalent combine it with another movement:  e.g. if I pitch my head up towards the camera or add a little bit of head roll before yawing.
When occurring, the yaw value will bounce between ~30 and ~40 degrees.

I use a PS3 Eye camera.

searching46dof, 2022-07-25:
this is an issue with correct_rotation attempting to compensate for yaw and pitch with the face being off center from the center of the preview window. when compensating it also needs to handle boundary conditions where the face is perpendicular to the direction of the camera

paedru, 2022-10-04:
Finally found the problem..it's has nothing to do with version..just reconnect the device i.e cam or apple phone sending the inputs that is connected via your pc ip address