AIRLegend/aitrack #75: Portrait mode for phones

iAmEMT, 2020-11-09:
I've been playing around with my iPhone 7+ and I noticed that the phone -> PC app (iVCam) I use lets me change from landscape to portrait mode, but using AItrack with iVCam in portrait mode does not work.

It would be great if we had the option to choose which format to use in the options since I currently don't have a stable way to mount my phone for landscape.

AIRLegend, 2020-11-09:
You should be able to do that by flipping the values under resolution in AITrack's settings (ie.if it is 1280x720, set 720x1080).

closed 2020-11-12 by AIRLegend

blueshack112, 2021-09-07:
> You should be able to do that by flipping the values under resolution in AITrack's settings (ie.if it is 1280x720, set 720x1080).

This works a bit weird for me. Consider below:
[image]

By default, I don't know if my camera has a weird zoom (C270 logitech 720p) or what but as soon as my beard or the top of my hair got cut out in the preview, the app would cut tracking. Since I have a big face, a lot of trackable area is possible but I suppose the pipeline cuts off when it doesn't detect a full head.

I tried to circumvent this by switching to portrait which would keep my full face and head in the frame for closer distance. Flipping the resilution just did this, the image above. However, suddenly, the application can now detect my face even when the head is cutout. So I'm guessing that this letterbox kinda thing only happened in the preview while the original frame that is being processed is good to go. Awesome. Just thought I should put it out in case you guys might need this data. Love the work. Absolutely stoked to see this working.

Some Reference:
With landscape mode point where I lose tracking:
[image]

With portrait mode config: (The track was solid but the preview was messed up)
[image]

All in all, this is not a bug for me.