AIRLegend/aitrack #76: Dropping frames at higher resolutions

1234filip, 2020-11-15:
**Describe the bug**
The software slows way down on higher resolutions.

**To Reproduce**
Works really well on 640x480.
When I try to bump it up to 1280x720 it introduces high latency and starts dropping frames.

 - Webcam model: Some generic Full HD webcam.
 - AITrack version: 0.6.3

**Additional context**
The CPU usage doesn't even max out, stays at about 10%. Also checked the cores, no core is maxed out. Memory usage seems to stay the same. I have a pretty good PC so running a face detection model shouldn't too hard for it. I think there a problem with loading higher resolution frames as the Fast/Heavy model difference in performance is non-noticable.

AIRLegend, 2020-11-16:
Maybe the problem is on the camera. 720p have been pretty tested and it seems to be okay. Could you have other process running and interfering with the capture?

1234filip, 2020-11-17:
It was a weird bug with the drivers of my camera. Fixed now.

closed 2020-11-17 by 1234filip