AIRLegend/aitrack #78: Program won't open when using PS3 eye

1234filip, 2020-11-17:
**Describe the bug**
The program won't open when using the PS3eye cam. Works on v0.6.2.

**To Reproduce**
Steps to reproduce the behaviour:
1. Plug in the PS3eye
2. Try to open the program

**Environment (please complete the following information):**
 - Webcam model: PS3 eye
 - AITrack version: 0.6.3

**Additional context**
The program loads for a couple of seconds and then closes itself. If I try to start it with the cam unplugged works fine but then won't detect the camera.

AIRLegend, 2020-11-19:
I have a PS3 myself and runs with no problem. Can you share your logfile? Also, you could try changing your camera to another USB port.

1234filip, 2020-11-19:
I found the problem. Last time I opened the program I chose a camera with an ID of 2. I then unplugged all the cameras, leaving only Camera 0. The program crashed because it was trying to access Camera 2 from the previous config. You should fix that :)

closed 2020-11-21 by AIRLegend