AIRLegend/aitrack #80: AITrack with DroidCam glitches out like crazy

Jackaroo05, 2020-12-11:
**Describe the bug**
Using DroidCam with AITrack produces glitched camera feed - either 2 feeds side by side that AITrack cannot track (theres also a big black area below the two feeds that you cant get rid of because the app will crash if you decrease the height too much) or it looks like a glitch in the matrix where the feed is skewed off to the right and someone put a glitch filter on top of it.

**To Reproduce**
Steps to reproduce the behavior:
I don't know. It started like this and even after a windows reinstall its still happening.

**Expected behavior**
Glitch-Free webcam feed when tracking

**Screenshots**
If applicable, add screenshots to help explain your problem.
Will try to get screenshots - but I do not have access to my computer for a while.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 10 version 20H2
 - Webcam model: Moto Z4 with DroidCam
 - AITrack version: 0.6.3

**Additional context**
Add any other context about the problem here.
Also, if you lower the resolution to fix the issue past a certain value (I cant remember) the app instantly crashes when attempting to start tracking.

AIRLegend, 2020-12-12:
Hi! Are you sure both output Droidcam resolution and AITrack resolution match? That issue is produced when the two aspect ratios are different.

Jackaroo05, 2020-12-12:
I ran it today and made sure the resolutions matched and it worked just fine. I don't know what caused it but i'm not going to question it. It is 2020 after all :) Ill be sure to come back if any issues arrive.

closed 2020-12-12 by Jackaroo05