AIRLegend/aitrack #85: Black screen using Droidcam

ShadowOps, 2021-01-17:
Droidcam is setup right and i get the camera feedback in my browser but not in AITrack, its just shows a black screen and nothing else. How to setup this right?

AIRLegend, 2021-01-17:
Hi,
Please, make sure that you're selecting the right input (Maybe you have "Camera 0" and there's another "Camera 1", for example). Also, check AITrack has Camera permissions (under Windows privacy settings) and no antivirus nor program is blocking it.

Hope that works :)

ShadowOps, 2021-01-20:
> 
> 
> Hi,
> Please, make sure that you're selecting the right input (Maybe you have "Camera 0" and there's another "Camera 1", for example). Also, check AITrack has Camera permissions (under Windows privacy settings) and no antivirus nor program is blocking it.
> 
> Hope that works :)

Yes, that did work! Had to select camera 2 or 3. Thanks for creating this amazing program.