AIRLegend/aitrack #90: Mouse emulation does not work

ErfolgreichCharismatisch, 2021-02-12:
I am using the most current release of opentrack with aitrack.
Input UDP over network.
Output mouse emulation.
The squid is moving according to my head movement.
But the mouse is not moving.
Neither with Legacy nor with directinput.
Also, once I start a game, here metro last light, aitrack turns off by itself.

AIRLegend, 2021-02-12:
Hi!

The mouse emulation part, is part of Opentrack. AITrack only sends data to it. Maybe you should post an issue on Opentrack's issues page :)

Regarding the AITrack crash. Any log/error?

ErfolgreichCharismatisch, 2021-02-12:
Where is aitracks log?

AIRLegend, 2021-02-12:
Next to the `.exe`

ErfolgreichCharismatisch, 2021-02-12:
Nothing special:

```
[2021-02-12 18:13:53.187] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2021-02-12 18:13:54.408] [aitrack] [info] Created/Found prefs.ini
[2021-02-12 18:13:54.409] [aitrack] [info] Searching for cameras...
[2021-02-12 18:14:00.813] [aitrack] [info] Number of recognized cameras: 2
[2021-02-12 18:14:00.813] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2021-02-12 18:14:00.813] [aitrack] [info] Building Tracker with selected camera: 0
[2021-02-12 18:14:00.947] [aitrack] [info] Tracker initialized.
[2021-02-12 18:14:00.947] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:14:00.947] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:14:00.947] [aitrack] [info] Updated camera parameters. 640x360@30
[2021-02-12 18:14:00.947] [aitrack] [info] Checking for updates
[2021-02-12 18:14:01.021] [aitrack] [info] App initialized
[2021-02-12 18:14:02.611] [aitrack] [info] Saving prefs
[2021-02-12 18:14:02.611] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:14:02.611] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:14:02.611] [aitrack] [info] Updated camera parameters. 640x360@30
[2021-02-12 18:14:02.611] [aitrack] [info] Tracker initialized.
[2021-02-12 18:14:02.613] [aitrack] [info] Prefs saved
[2021-02-12 18:14:12.319] [aitrack] [info] Saving prefs
[2021-02-12 18:14:12.319] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:14:12.319] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:14:12.319] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:14:12.319] [aitrack] [info] Tracker initialized.
[2021-02-12 18:14:12.321] [aitrack] [info] Prefs saved
[2021-02-12 18:14:18.829] [aitrack] [info] Saving prefs
[2021-02-12 18:14:18.829] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:14:18.829] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:14:18.829] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:14:18.829] [aitrack] [info] Tracker initialized.
[2021-02-12 18:14:18.832] [aitrack] [info] Prefs saved
[2021-02-12 18:14:20.386] [aitrack] [info] Starting camera 1 capture
[2021-02-12 18:14:21.170] [aitrack] [info] Camera 1 started capturing
[2021-02-12 18:14:23.361] [aitrack] [info] Stop camera 1 capture
[2021-02-12 18:14:27.484] [aitrack] [info] Saving prefs
[2021-02-12 18:14:27.484] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:14:27.484] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:14:27.484] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:14:27.484] [aitrack] [info] Tracker initialized.
[2021-02-12 18:14:27.485] [aitrack] [info] Prefs saved
[2021-02-12 18:14:30.394] [aitrack] [info] Starting camera 0 capture
[2021-02-12 18:14:31.461] [aitrack] [info] Camera 0 started capturing
[2021-02-12 18:14:35.577] [aitrack] [info] Stop camera 0 capture
[2021-02-12 18:15:11.951] [aitrack] [info] Saving prefs
[2021-02-12 18:15:11.951] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:15:11.951] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:15:11.951] [aitrack] [info] Updated camera parameters. 640x360@30
[2021-02-12 18:15:11.951] [aitrack] [info] Tracker initialized.
[2021-02-12 18:15:11.956] [aitrack] [info] Prefs saved
[2021-02-12 18:15:14.326] [aitrack] [info] Starting camera 1 capture
[2021-02-12 18:15:15.170] [aitrack] [info] Camera 1 started capturing
[2021-02-12 18:15:21.417] [aitrack] [info] Stop camera 1 capture
[2021-02-12 18:15:26.272] [aitrack] [info] Starting camera 1 capture
[2021-02-12 18:15:27.108] [aitrack] [info] Camera 1 started capturing
[2021-02-12 18:15:31.768] [aitrack] [info] Stop camera 1 capture
[2021-02-12 18:15:40.290] [aitrack] [info] Saving prefs
[2021-02-12 18:15:40.290] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:15:40.290] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:15:40.290] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:15:40.290] [aitrack] [info] Tracker initialized.
[2021-02-12 18:15:40.292] [aitrack] [info] Prefs saved
[2021-02-12 18:15:44.235] [aitrack] [info] Saving prefs
[2021-02-12 18:15:44.235] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:15:44.235] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:15:44.235] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:15:44.235] [aitrack] [info] Tracker initialized.
[2021-02-12 18:15:44.238] [aitrack] [info] Prefs saved
[2021-02-12 18:15:46.180] [aitrack] [info] Starting camera 0 capture
[... 20 lines cut ...]
[2021-02-12 18:32:25.219] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:32:25.219] [aitrack] [info] Rebuilding tracker with new parameters
[2021-02-12 18:32:25.297] [aitrack] [info] Tracker initialized.
[2021-02-12 18:32:25.298] [aitrack] [info] Prefs saved
[2021-02-12 18:32:27.721] [aitrack] [info] Starting camera 0 capture
[2021-02-12 18:32:28.667] [aitrack] [info] Camera 0 started capturing
[2021-02-12 18:32:36.552] [aitrack] [info] Stop camera 0 capture
[2021-02-12 18:32:39.362] [aitrack] [info] Saving prefs
[2021-02-12 18:32:39.362] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:32:39.362] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:32:39.362] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:32:39.362] [aitrack] [info] Rebuilding tracker with new parameters
[2021-02-12 18:32:39.431] [aitrack] [info] Tracker initialized.
[... cut ...]
[2021-02-12 18:32:42.186] [aitrack] [info] Camera 0 started capturing
[2021-02-12 18:33:27.034] [aitrack] [info] Saving prefs
[2021-02-12 18:33:27.034] [aitrack] [info] Updated stabilizer.
[2021-02-12 18:33:27.034] [aitrack] [info] Updating camera parameters...
[2021-02-12 18:33:27.034] [aitrack] [info] Updated camera parameters. 1920x1080@30
[2021-02-12 18:33:27.034] [aitrack] [info] Tracker initialized.
[2021-02-12 18:33:27.039] [aitrack] [info] Prefs saved
[2021-02-12 18:33:44.190] [aitrack] [info] Stop camera 0 capture

```

AIRLegend, 2021-02-13:
Doesn't seem like it crashed 🤔

ErfolgreichCharismatisch, 2021-02-13:
I fgured the solution in opentrack. Now, my input is 1920 x 1080, which makes detection slow. Any idea about how to get a smaller resolution for detection?

AIRLegend, 2021-02-13:
The resolution must be the same as your camera. If your camera doesn't support going to other resolutions it'll probably crash.

Sorry, but the models are trained to recognize a normal face and therefore it's not meant for recognizing an ocluded one :(. I suggest you taking a look at the aruco tracker built in Opentrack :).

AIRLegend, 2021-02-14:
Closing because this isn't related with AITrack. Feel free to come by the Discord server if you have any further questions :).

closed 2021-02-14 by AIRLegend