AIRLegend/aitrack #92: CTRL+T Shortcut options

Dymerz, 2021-02-27:
**Describe your idea**
Add an option to enable/disable the CTRL+T shortcut, I'm always frustrated when I'm want to open a new tab (CTRL+T) in my web explorer and AITrack catch the event, so my new tab isn't opened.

**Describe the solution you'd like**
An option to enable/disable the CTRL+T shortcut.

**Describe alternatives you've considered**
A simple boolean in the prefs.ini or the user could choose which key to use for this feature in the ini/GUI.

gabrielbunselmeyer, 2021-04-29:
+1 to this idea. It's pretty annoying currently 😬

UNDEFINED-BEHAVIOR, 2021-08-08:
Hello
Recently it's been annoying enough that I've implemented shortcuts toggling myself. It's "works on my machine" status right now but if you don't mind I could arrange a pull request perhaps.
[image]

AIRLegend, 2021-08-08:
Hey @UNDEFINED-BEHAVIOR, cool! Glad to merge it if you don't mind making the PR :). 

Could you move it into the configuration menu though? The main view should be as clean as possible.

UNDEFINED-BEHAVIOR, 2021-08-08:
I'll think about it, will just make the PR as is first.
If it's in config menu then it should also go in to pref.ini

UNDEFINED-BEHAVIOR, 2021-08-09:
[image]
Ok its been moved to config window, and should be saved in pref file now.

AIRLegend, 2021-08-09:
Ok, this is merged and it will be available with the next version :) Thank you very much @UNDEFINED-BEHAVIOR!

For those who want this right now, you have the compiled program here (Hope you can access it): https://ci.appveyor.com/project/AIRLegend/aitrack/builds/40302101/artifacts

Closing. Have a nice day :)

closed 2021-08-09 by AIRLegend