AIRLegend/aitrack #94: Camera disappears after first use

ExTatum, 2021-04-13:
**Describe the bug**
Let me start off by saying that I've first made sure my camera worked through the windows "Camera" app. I downloaded the installer and after first running the tracking, the preview would start twitching and mirroring, then turn pitch black after a few seconds and the camera wouldn't be detected anywhere (devices manager, camera, AITrack), only to reappear whenever I unplug the USB port and plug it back in. Interestingly enough the light remains on until I unplug it, despite closing all programs using it down. Now whenever I try to troubleshoot the problem through the camera app, the camera will "disappear" as soon as it detects a face (after device restart from safe mode it showed a tolerance of a few seconds between detecting my face and disappearing, though consecutive tries resulted in the webcam disappearing even at slight sudden movements (waving my hand))

**To Reproduce**
Steps to reproduce the behavior:
1. Open AITrack
2. Click on Start Tracking
3. Scroll down to '....'
4. See error

**Expected behavior**
lack of twitching and overall crashing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 10 home, 2004 (OS Build: 19041.867)
 - Webcam model: HD Webcam C270 (logitech)
 - AITrack version:  v0.6.4-alpha

**Additional context**
none

ExTatum, 2021-04-13:
Found a temporary fix to the problem. If you start the tracking with the camera plugged in, then unplag and replug it (don't stop the tracking) and then juggle through detected cameras in the configuration menu, stopping and restarting the tracking every time you swap you can temporarily get your camera back. I got it to work for the camera app as well though I don't remember how

ExTatum, 2021-04-13:
Soo.... This just got wilder. The problem seems to be fixing itself whenever I unplug a usb adopter I use for my headphones (razer electra) and microphone (sf-666)

AIRLegend, 2021-04-13:
Seems like a problem related to your setup (USB adapters, drivers, ...). That camera model is reportedly to be working okay. 

Have you tried connecting it directly to different USB ports? (i.e. not via hubs/adapters). Maybe USB3 ones?

ExTatum, 2021-04-15:
the camera was connected direcly to the pc via usb anyway. What I unplugged was another usb in the port next to it . Keep in mind that it was working perfectly before using AItrack so it must be the root of the problem