AIRLegend/aitrack #96: PS3 Eye Camera wont turn on

22userman, 2021-04-17:
**Describe the bug**
Aitrack v6.4 doesn't turn on the ps3 eye camera.
**To Reproduce**
Steps to reproduce the behavior: Using a ps3 eye camera
1. Start aitrack
2. Click on 'start tracking'

**Expected behavior**
See image from camera

**Screenshots**
See attached.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 10 Home 20H2
 - Webcam model: PlayStation 3 Eye Camera
 - AITrack version: AiTrack\aitrack-v0.6.4-alpha

**Additional context**
The CL-Eye Test displays an image with the ps3 eye camera.
Aitrack works correctly if I use my Microsoft LifeCam Cinema 720p HD camera.
[image]
[image]
[image]

AIRLegend, 2021-04-18:
Strange. Try deleting `prefs.ini` before running `AITrack.exe`. Maybe it's  related to your drivers.  Check whether using these solve the problem: https://zadig.akeo.ie/. 
For sure the PS3 Camera must work without any problem.

asch75, 2021-04-24:
Same problem here. All is working fine with my logitech C925e, but my pS3 Eye with CL-Eye driver is not working.
Just black screen.
Going to test Zadig.

AIRLegend, 2021-04-24:
Grea! Closing.

closed 2021-04-24 by AIRLegend