AIRLegend/aitrack #99: AITrack Silently Crashes Upon hitting "Apply" in configuration settings

DavidJBloom, 2021-04-27:
**Describe the bug**
AITrack v.0.6.4-alpha opens when run as administrator with no issue, displays "No Active Video Feed". 

If "Start Tracking" is chosen, camera "on" light will blink a few times and either the image will appear, or the program will silently crash. 

If "Configuration" is chosen, configWindow will open with all settings modifiable. Then, regardless of modification of other setting, if you hit apply, AiTrack will silently crash. Repeat ad nauseum. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configuration'
2. Click on 'Apply'
4. AiTrack Silently Crashes.

**Expected behavior**
Upon hitting "Apply", any settings changed should be applied and configWindow should stay in place until closed directly, leaving only the main window open and ready for tracking start. 

**Environment (please complete the following information):**
 - Operating system version (Windows 10 Home 1909)
 - Webcam model: GESMA - https://tinyurl.com/tvfdws6z
 - AITrack version: v.0.6.4-Alpha

**Additional context**
I have tried uninstalling and reinstalling, the same error occurs. 

Thanks for your help.

AIRLegend, 2021-04-29:
Sounds like AITrack hasn't write permissions.

AIRLegend, 2021-09-24:
Closing due to inactivity.

closed 2021-09-24 by AIRLegend