AIRLegend/aitrack #100: High resolution is quite buggy

Jacktherookie, 2021-04-29:
**Describe the bug**
A clear and concise description of what the bug is.
When I'm trying to change resolution to FHD (1920x1080) in Settings, it really buggy.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings
2. Change resolution to FHD (1920x1080)
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Camera fps drop down to like 2 fps

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): 
 - Webcam model: 
 - AITrack version:

**Additional context**
Add any other context about the problem here.

AIRLegend, 2021-04-29:
Most cameras reduce their FPS on low-light settings. Could that be your case?

If your camera supports a lower resolution you should use it because a higher one will only eat more CPU. Internally, the program downscales the video feed either way.

Jacktherookie, 2021-04-30:
I dont think low light is a problem for me. I'm pretty sure I have decent light for my webcam. But I did reset my PC and It work smoothly. I think something in my PC was hindering Track.

closed 2021-04-30 by Jacktherookie