AIRLegend/aitrack #102: AI Track app crashes upon clicking "Start Tracking"

CowboyCoastie, 2021-05-15:
**Describe the bug**
Application silently crashes after "Start Tacking" button is clicked.

App worked as expected upon first loadup, aside from camera being positioned too far back in cockpit of Cessna Citation CJ4. Rebooted app in attemp to correct camera location and not app won't even run without crashing.

**Expected behavior**
Click Start Tracking
Camera lights up
Face is recognized
Camera is placed in cockpit correctly

**Environment (please complete the following information):**
 - Operating system version (Windows 10 Home 20H2): 
 - Webcam model: 
 - AITrack version: 0.6.4-Alpha

AIRLegend, 2021-05-16:
What's the content of the log file?

CowboyCoastie, 2021-05-16:
Where do I find the log file?

Sent via the Samsung Galaxy S10, an AT&T 5G Evolution capable smartphone

CowboyCoastie, 2021-05-16:
> What's the content of the log file?

[2021-05-16 13:03:14.407] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2021-05-16 13:03:15.152] [aitrack] [info] Created/Found prefs.ini
[2021-05-16 13:03:15.152] [aitrack] [info] Searching for cameras...
[2021-05-16 13:03:16.578] [aitrack] [info] Number of recognized cameras: 1
[2021-05-16 13:03:16.580] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2021-05-16 13:03:16.580] [aitrack] [info] Building Tracker with selected camera: 0
[2021-05-16 13:03:16.720] [aitrack] [info] Tracker initialized.
[2021-05-16 13:03:16.720] [aitrack] [info] Updated stabilizer.
[2021-05-16 13:03:16.720] [aitrack] [info] Updating camera parameters...
[2021-05-16 13:03:16.720] [aitrack] [info] Updated camera parameters. 635x480@30
[2021-05-16 13:03:16.720] [aitrack] [info] Checking for updates
[2021-05-16 13:03:16.797] [aitrack] [info] App initialized
[2021-05-16 13:03:23.622] [aitrack] [info] Saving prefs
[2021-05-16 13:03:23.622] [aitrack] [info] Updated stabilizer.
[2021-05-16 13:03:23.622] [aitrack] [info] Updating camera parameters...
[2021-05-16 13:03:23.623] [aitrack] [info] Updated camera parameters. 635x480@30
[2021-05-16 13:03:23.623] [aitrack] [info] Tracker initialized.
[2021-05-16 13:03:23.659] [aitrack] [info] Prefs saved
[2021-05-16 13:03:27.927] [aitrack] [info] Starting camera 0 capture
[2021-05-16 13:03:29.069] [aitrack] [info] Camera 0 started capturing

AIRLegend, 2021-05-22:
That camera resolution seems odd. That's probably the issue.

CowboyCoastie, 2021-05-22:
What do you suggest it be changed to?

AIRLegend, 2021-05-23:
I don't know your specific camera model, but common resolutions are: 640x480, 1280x720 or 1920x1080. You should search for it though.

Fracking, 2021-05-24:
Hi,
in my case I fixed the crash by manually setting the fps. For me: 60. Seems to work now. :)