AIRLegend/aitrack #105: AITrack won't run on my PC

canuckdave, 2021-06-16:
**Describe the bug**
When I click on aitrack.exe I get the following message 

[image]

.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 10 Pro 10.0.19043
 - Webcam model: Droidcam
 - AITrack version: v0.6.4

AIRLegend, 2021-06-16:
This isn't a bug. It's simply related to your specific Windows configuration.

Maybe this could help you: https://windowsreport.com/app-cant-run-windows-10/#:~:text=Disable%20SmartScreen,(turn%20off%20Windows%20SmartScreen).

canuckdave, 2021-06-17:
Thanks for taking the time to reply and provide the link. I tried a number of the suggestions in the article but none worked.  However I think I now know what the issue is.  My Windows 10 is 32 bit and I suspect the app is 64 bit.  Also apologies for using the bug label.  As a new user of github I didn't realize there were other labels available.  Your revision to the help wanted label is more appropriate.

AIRLegend, 2021-06-17:
Don't worry :)

Yeah, the program is compiled only for 64bit as it's the most common architecture right now. So, probably, it's due to that...

Sorry about the inconvenience :/

closed 2021-07-18 by AIRLegend