AIRLegend/aitrack #107: Getting big pitch input when performing yaw right

sramekj, 2021-06-20:
Hi,

I am using AITrack v0.6.4-alpha with Opentrack-2.3.13.

My camera is Logitech c922 connected to USB 3.0

Cam settings: mounted slightly above my head, dist. 0.3m, 1280x720, 60fps, 70 FOV, no custom brightness, model type: fast, Landmark stabilization: on

The issue:

I use inverted Z and Pitch axis in opentrack (because of DCS).

X,Y,Z Pitch and Roll are working perfectly (when I test the inputs "standalone").

the problem is with Yaw. When I turn head left it performs well, however when I turn my head to the right it gets mixed with huge pitch up (in the actual game it is pitch down - because of inverted pitch axis) input.

I have tried it without inverted axis but the issue prevailed. I have also tried to disable roll axis in opentrack but this did not help as well.

Could you help me  please?

Codebreakerblue, 2022-04-18:
I have the same problem, with nearly the same setup. AITrack v0.6.5-alpha, opentrack-2022.1.0.

Camera is an ONN brand 1440p "Webcam with Autofocus" over USB 2.0 (being used at 640x480 for tracking,) 85 FOV/30fps, about 30cm away and ~5cm above, model type: medium/heavy, landmark stabilization on

My pitch is inverted, and XYZ are disabled.

My face is lit from the side by a lamp with a diffuse shade. Yawing my head towards the light source induces a massive roll response up to the 180 degree rotation limit of opentrack. When yawing my head the other direction, there is a massive pitch down (up but inverted) response instead. The AITrack preview shows nothing being amiss, and even tracks the mostly shadowed side of my face without issue.

Putting the light source directly above my webcam, the roll response issue starts happening on *both* yaw directions instead of just one. Disabling the roll response entirely does fix this, but it's not perfect because you lose all roll input.

searching46dof, 2022-06-25:
I noticed there is a opentrack-2022.3.0

I'm using aitrack v0.6.5alpha with opentrack-2022.2.0 and have very good results after lots of testing (3 weeks) on a Ryzen 5700g so the following info may help:
PS3 Eye Cam at 640x480x30fps (60fps causes too much jitter) zoomed in (56deg FOV) to reduce my background.
model type = heavy and landmark stabilization on to reduce jitter.
gain and exposure to 2/3 max (0=left) which is a little overexposed to help face detection.
I'm approx 4 meters away from the tv with the camera above the tv and a central ceiling light. In another room on a 2nd system with a corner torch lamp, I cannot place the camera above the tv due to detection problems. With the indirect light from the ceiling, I am limited to approx +20deg up but have no issue looking down.

I am using opentrack to map the settings for DCS simulating being strapped into the cockpit.
for yaw, +/-30deg looking at the left and right side of the tv maps to a maximum of +/- 120deg equivalent to looking over the shoulder. there is a +/-3deg dead zone near the center (less than 3 deg is mapped to zero) to reduce jitter.
for pitch, +/-30deg looking at the top and bottom of the tv maps to a maximum of +/- 90deg equivalent to looking up and down. there is a +/-3deg dead zone near the center.
for roll, +/-30deg maps to a maximum of +/-30deg. there is a +/-3deg dead zone by the center.
I disable x and y since I am strapped into the cockpit and I don't need to worry about being centered in the cockpit when I sit for a new game.
for z, +/-15cm maps to a maximum of +/-15cm since I am strapped in. there +/-3cm dead zone by the center.
I'm inverting pitch and z on the output.
Accela is used as the filter with both rotation and position filtering at maximum to reduce jitter.

looking horizontal or up (left, center and right), my head is against a headrest so there is no jitter problem.
looking down, my head isn't against the headrest but the jitter is minimal. pitch is scaled 3x but I believe my head is in a more stable rested position.
looking down to the left or right at the side instrument panels, there is too much jitter. pitch is scaled 3x and yaw is scaled 4x. I believe this is due to lack of head/neck support since my head is not against a headrest and my distance of 4m from the camera magnifying the jitter.

searching46dof, 2022-07-25:
there is an issue with solve_rotation.
for now try to center your face in the preview window to reduce the yaw and pitch compensation from being off center and use opentrack to limit the input from aitrack yaw and pitch to +/-30degrees

DrJaymz, 2022-12-05:
when I updated to latest version I now have this problem.  When I look left and right YAW I get huge ROLL inputs.  It just doesn't work anymore for the purposes I was using it for.

I thought there was a problem with it getting the axes mixed up but they are all fine.  x,y,z are all fine, reset profile nope.

searching46dof, 2022-12-06:
@DrJaymz, what version of aitrack/opentrack are you using? there was a coordinate swapping issue that was fixed in v0.6.6

I verified aitrack 0.7.0 and opentrack 2022.3.2 are tracking  x, y, z, yaw, pitch and roll coordinates correctly.