AIRLegend/aitrack #108: Cannot start AItrack.exe

Rikum0, 2021-07-29:
**Describe the bug**
The software will not even start. I _sometimes_ see it appear for less than a second on my task bar. It does not even appear in my task manager. Adding Antivirus exceptions and administrator authorizations, and reinstalling Visual C++ did not change anything.

**Expected behavior**
Starting like a normal software

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): 
 - Webcam model: 
 - AITrack version: v0.6.4

AIRLegend, 2021-07-31:
Could you test 0.6.4 and paste the contents of `log.txt`?

Rikum0, 2021-07-31:
Where can I find the logs? If they are supposed to appear in the software's file, I think it crashes before it can even create a log.

AIRLegend, 2021-08-08:
Are you sure it has write permissions? Try executing as admin, moving to desktop or checking Avast isn't blocking it. Creating the log is literally the first thing it does.

AMKaxel, 2021-09-14:
I am having the same issue, but with version 0.6.5 I am not sure what is happening, I checked the task manager but nothing, it also doesn't has a log. I am starting to think it is having some compatibility issue, or some program is interfering with it.

AMKaxel, 2021-09-14:
I solved !!.
I turned off Avast and it worked, which is freaking me out a bit, if I have to disable it, there must be some thing that Avast doesn't likes. 
SuS

AMKaxel, 2021-09-14:
UPDATE. I have found the source of the problem. It is the Log it self. 
The Avast Anti-Ransomware module blocks it. 
This happens when you put the download folder that is protected by the Anti-Ransomware module, the explanation is simple, the module detects a program trying to modify a file in a "protected" folder, this triggers it and course it to do its job and block the action. This also explains why there is no log file in the folder, because creating it triggers the module that kills the action. 

I'm not saying that this program is a Ransomware, I have scanned the files with over a hundred Anti-Virus softwares and non found any thing wrong with the program. You can download it safely.

AIRLegend, 2021-09-24:
I'm closing this issue them.

If anyone is facing this issue it's probably related to your antivirus or because `AITrack.exe` has no write permissions.

closed 2021-09-24 by AIRLegend