AIRLegend/aitrack #109: AI for short distances

duo1204, 2021-07-31:
**Describe your idea**
my eyes to webcam distance is about 20cm and it seems no matter what parameter 
AItracker only works well with 1m distance.
my room is well rit using a ramp

**Describe the solution you'd like**
don't know anything about ai or how you did it but could it be possible to allow users to select short distance?

**Additional context**
ai just doesn't recognize well in short distance i think

AIRLegend, 2021-07-31:
Hi,
Your full face needs to be within the viewport. If not, the face won't be detected. 
In case your full face is visible, then, the problem should be in the position estimation part. I'm experimenting with that.

blueshack112, 2021-09-07:
Hey,

This might be of use to you. I faced the same issue with the same webcam, I think (mine is c270). Rotating to portrait (very longer height, shorter width) allows your face to stay in the frame for very close distances.

I suppose this would give you a disadvantage if you want to move in x and/or y directions and roll. But your pitch, yaw, and z movement will track while being very close to the camera. The basic gist is that you want to make your frame bigger to accommodate more movement. I've noticed that even 1cm of the top of the head being cut off is enough for AITrack to cut tracking off completely.

I am no expert, just my observations. Hopefully, a dev team member can verify this.

https://github.com/AIRLegend/aitrack/issues/75#issuecomment-914145499

searching46dof, 2022-07-15:
AIRLegend's remarks regarding the full face needs to be within the viewport seem to be a requirement. It's searching for the borders of a head indicated by the pink dots in the viewport.

Note that there are issues initializing head3dScale and camera_matrix and a logic issue in correct_rotation contributing to desensitization at short distances. The OPTIMIZE_PositionSolver fixes at https://github.com/searching46dof/aitrack/blob/master/AITracker/src/PositionSolver.cpp allow tracking at distances shorter than 1meter. I've tested the changes at less than 0.5 meter where my head almost fills the entire viewport. When tracking at this close range, the FOV and distance need be fairly accurate. If the distance is inaccurately set for 3 meters, then it will easily lose face tracking when looking around.