AIRLegend/aitrack #110: Aitrack not responding

Adamantyl, 2021-08-02:
Upon starting the program (v. 0.6.4 alpha) a blank window pops up but the program is not responding. I have to force quit the program. The same issue happens in 0.6.3 alpha. I am using Windows 10 Home edition. My camera is a logitech C270, but the bug occurs when the camera is plugged in or not. 

**Screenshots**
[image]

AIRLegend, 2021-09-24:
Hey, sorry for the delay. Does 0.6.5 solve this?

searching46dof, 2022-06-30:
If the camera is plugged directly to the computer's usb port, also try connecting it indirectly via a powered usb hub.

raeldor, 2024-08-14:
I have the same problem.