AIRLegend/aitrack #113: Glitchy behavior at 20 degrees or more PITCH axis (direction depends on MODEL type)

saayanbiswas, 2021-08-18:
**Describe the bug**
This bug makes the facetrack move all over the place back and forth rotate and stuff. It occurs usually on 20 degrees of raw input 
of PITCH. Fascinating stuff is This bug is reversed on different models(fast, medium, heavy). 

**To Reproduce**
(all pitch values are RAW inputs shown in opentrack)
1. On the FAST model the irregular behavior is visible at looking ABOVE at 20 degrees of pitch, while it is completely normal while looking down
2. On the MEDIUM and HEAVY model the irregular visible is visible at nearly 20 degrees of pitch looking DOWNWARDS where looking upwards is  fine.
3. Another Fascinating stuff is that the Facetrack works normal once it moves over 25-30 degrees of pitch, indicating lack of face reading at specific angles, also depending on which mode it is on. 

**Expected behavior**
The tracker should rotate smoothly through all the angles (progressive accuracy deterioration is justified)
(but it glitches at 20-25 degrees of pitch and restores after )

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number):  Windows 10 Version 20H2(Build 19042.1165)
 - Webcam model: Dell G3  3579 Integrated Webcam 
 - AITrack version: v0.6.4 -alpha

**Additional context**
PITCH is where the bug lies. Yaw, roll, translation all works flawlessly.

SmokeSwitch, 2021-08-19:
I have exactly the same issue (with a no name USB webcam and the latest version of AITrack)

saayanbiswas, 2021-08-29:
I had plans to test the previous version for the bug. I couldn't test until today.
I have 0.6.4, 0.6.3 and  0.6.2 versions.
Bug present in 0.6.4
Bug present in 0.6.3
 I was UNABLE to recreate the bug in 0.6.2.  seems like the version 0.6.2 doesn't have this specific bug. 
 which suggests that this bug recently developed due to some changes from 0.6.2. If i am Correct the major change about taking consideration about Camera FOV may have a link with this since it was introduced in 0.6.3. Please look into this matter

saayanbiswas, 2021-08-29:
> **Describe the bug**
> This bug makes the facetrack move all over the place back and forth rotate and stuff. It occurs usually on 20 degrees of raw input
> of PITCH. Fascinating stuff is This bug is [...]

EDIT:
I had plans to test the previous version for the bug. I couldn't test until today.
I have 0.6.4, 0.6.3 and  0.6.2 versions.
Bug present in 0.6.4
Bug present in 0.6.3
 I was UNABLE to recreate the bug in 0.6.2.  seems like the version 0.6.2 doesn't have this specific bug. 
 which suggests that this bug recently developed due to some changes from 0.6.2. If i am Correct the major change about taking consideration about Camera FOV may have a link with this since it was introduced in 0.6.3. Please look into this matter

closed 2021-08-29 by saayanbiswas